riscv/riscv-arch-test #146: Test I-ECALL-01.S

Imperas, 2020-12-02:
This test writes values to the MTVEC register and assumes that all bits are writable.
this is not a requirement of the RISCV spec, and some bits of the MTVEC register can be omitted in the case where the exception handlers are expected to be aligned to a certain byte boundary

Additionally, this test assumes that the DEBUG mode is not implemented on the target device, and that an ECALL instruction will cause an exception to occur, rather than a mode switch to DEBUG mode

I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

pdonahue-ventana, 2020-12-02:
> Additionally, this test assumes that the DEBUG mode is not implemented on the target device, and that an ECALL instruction will cause an exception to occur, rather than a mode switch to DEBUG mode

The ecall instruction never causes a mode switch to debug mode.  It would be bad if every ABI and/or SBI call had to be handled by the external debugger.  The only mention of "ecall" in the debug spec is to tell debuggers not to execute ecall while in debug mode.

The concern about mtvec is valid, though.

allenjbaum, 2020-12-02:
I believe the MTVEC problem will go away when the RFQ tests are pulled in. The new standard trap handler prolog will attempt to point MTVEC to the handler entr. If that fails, it will leave the current value unchanged, but replace the code that it points to with trampoline code. If MTVEC is not writable, and the code it points to is not writable, then the test will fail. 
IF this is inadequate, then we have two fallback plans: force trampoline alignment to 4K, which will probably fix most problems, or allow the model to supply a writable trampoline address that MTVEC can be set to.

Imperas, 2020-12-03:
So until resolved at some point in the future...
> I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

Do you not agree ?

allenjbaum, 2020-12-03:
No, I do not. These tests appear to wrk for the majority of implementations (based on the fact that only two have reported that they don't pass) and there is no reason to weaken the already weak testing to accommodate those implementations.
The consensus is that we will add a note into the README that indicates that the tests in the repo assume specific architectural options (specifically: MTVEC is completely writable, and that HW misalign is not supported) and that implementations that fail tests that assume the opposite should apply for a waiver when applying for the trademark.

allenjbaum, 2022-08-11:
will be addressed by new trap handler

pawks, 2023-02-09:
Addressed by PR #277

closed 2023-02-09 by pawks