riscv/riscv-arch-test #147: Test I-MISALIGN_LDST.S

Imperas, 2020-12-02:
This test writes values to the MTVEC register and assumes that all bits are writable.
this is not a requirement of the RISCV spec, and some bits of the MTVEC register can be omitted in the case where the exception handlers are expected to be aligned to a certain byte boundary

Additionally, this test assumes that the target does not implement hardware support for misaligned load/store accesses, and that these should raise a load/store misaligned exception, this is not the case for targets providing hardware support to resolve these in the bus interface unit.

I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

allenjbaum, 2020-12-02:
I believe the MTVEC problem will go away when the RFQ tests are pulled in. The new standard trap handler prolog will attempt to point MTVEC to the handler entr. If that fails, it will leave the current value unchanged, but replace the code that it points to with trampoline code. If MTVEC is not writable, and the code it points to is not writable, then the test will fail. 
IF this is inadequate, then we have two fallback plans: force trampoline alignment to 4K, which will probably fix most problems, or allow the model to supply a writable trampoline address that MTVEC can be set to.

The new trap handler also will deal with misaligned traps, and there are separate tests with separate .ref files for that.
Changing the makefrag that selects the tests might be all that is required to do this - we are looking into the best way to do it. Riscof does this automaticallly based on the YAML file, of course.

Imperas, 2020-12-03:
So until resolved at some point in the future...
> I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

Do you not agree ?

allenjbaum, 2020-12-03:
No, I do not. These tests appear to wrk for the majority of implementations (based on the fact that only two have reported that they don't pass) and there is no reason to weaken the already weak testing to accommodate those implementations.
The consensus is that we will add a note into the README that indicates that the tests in the repo assume specific architectural options (specifically: MTVEC is completely writable, and that HW misalign is not supported) and that implementations that fail tests that assume the opposite should apply for a waiver when applying for the trademark.

allenjbaum, 2020-12-04:
Also notice this is appears too be a duplicate of issue#22

Ubilite, 2020-12-10:
Concerning misaligned loads and stores, in larger memory architectures (for example 64/128bit memory), there could be a benefit in allowing for a mixture of misaligned accesses to be handled by HW and others by a misaligned interrupt.
I was wondering if that would not be RISCV compliant.
An example of this would be for when issuing a load half-word or store half-word with a byte index of 1 on 64b wide memory, it is very easy for HW to service that and may subsequently decide not to raise a misaligned exception there.
What is the view in terms of compliance for such architectures.

allenjbaum, 2020-12-11:
You are quite right. There is no way to describe that configuration in the
architecture that I know of.
I'm working on it, but it will require framework 2.0 to configure the
reference model (SAIL) to exhibit precisely that implementation-dependent
behavior and use it as the reference signature.

pdonahue-ventana, 2020-12-11:
The architecture description of PMA says:
> Access types specify which access widths, from 8-bit byte to long multi-word burst, are supported, and also whether misaligned accesses are supported for each access width.

and:
> Memory regions that support aligned LR/SC or aligned AMOs might also support misaligned LR/SC or misaligned AMOs for some addresses and access widths.

and:
> Implementations may raise access-fault exceptions instead of address-misaligned exceptions for some misaligned accesses, indicating the instruction should not be emulated by a trap handler.

So the configuration of the reference model needs to be able to, for instance:
- allow misaligned accesses for some access widths but not others within a particular PMA region.  I don't think that the architecture allows an implementation to execute a halfword load to address 1 but to trap a halfword load to address 3 (assuming addresses 0-7 are in the same PMA region).
- allow misaligned AMOs for certain addresses within a particular PMA region.  In theory, the set of allowed/disallowed misaligned addresses+widths can be arbitrary.  Normally, an implementation that supports misaligned AMOs might only support ones that don't span a power-of-two boundary (so this is probably what the config file should be able to support).
- specify the type of exception to be raised for a particular address+width.  Again, this could be an arbitrarily complex mapping but likely won't be in reality.

allenjbaum, 2020-12-11:
It’s worse than that. 
An implementation could trap only on accessed that crosses a cache line or page boundary.
It’s even worse than that- an implementation might trap if the second half of the access got a TLB miss- so sometimes it traps, and sometimes it doesn’t, for the same address.
 That is legal behavior. Not sure how tests can handle that.

-Allen

allenjbaum, 2022-08-11:
We will only test the deterministic case. Needs coordination with Sail.
Will file an issue to write self-checking tests in the future.

allenjbaum, 2023-02-22:
It is currently possible to configure tests to testr a DUT that always trapped on misaligned load/stores, or always succeeds in loading and storing misaligned stores (but not one that sometimes does and sometimes doesn't). The tests are only applied to memory address ranges, so we will not test IO (since tests can't know where those are, and even if they could, can't know the contents of the PMA). Since this should now work for the determinstic memory-only case, I'm closing this. IF it turns out not to work work, re-open the issue

closed 2023-02-22 by allenjbaum