riscv/riscv-arch-test #148: Test I-MISALIGN_JMP.S

Imperas, 2020-12-02:
This test writes values to the MTVEC register and assumes that all bits are writable.
this is not a requirement of the RISCV spec, and some bits of the MTVEC register can be omitted in the case where the exception handlers are expected to be aligned to a certain byte boundary

Additionally, this test assumes that the target does not implement the C extension, such that half word aligned fetches are legal, where a target does implement the C extension, then fetches to these address boundaries are legal, although the fetch of a 32bit instruction, will likely yield an illegal instruction exception, not an instruction address misaligned exception

I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

allenjbaum, 2020-12-03:
I believe the MTVEC problem will go away when the RFQ tests are pulled in. The new standard trap handler prolog will attempt to point MTVEC to the handler entr. If that fails, it will leave the current value unchanged, but replace the code that it points to with trampoline code. If MTVEC is not writable, and the code it points to is not writable, then the test will fail. 
IF this is inadequate, then we have two fallback plans: force trampoline alignment to 4K, which will probably fix most problems, or allow the model to supply a writable trampoline address that MTVEC can be set to.

 Riscof does this automaticallly based on the YAML file, of course, and we may want to wait for that to fix it.
 Riscof will also deal with the C-extension issue, another reason to wait for riscof. The IBEX test issues have been there since the model support files were put into the repo 16 months ago, so there appears to be no urgency.

Imperas, 2020-12-03:
So until resolved at some point in the future...
> I suggest this test is deprecated, until it can be enhanced to (at least) cover the above.

Do you not agree ?

allenjbaum, 2020-12-03:
No, I do not. These tests appear to wrk for the majority of implementations (based on the fact that only two have reported that they don't pass) and there is no reason to weaken the already weak testing to accommodate those implementations.
The consensus is that we will add a note into the README that indicates that the tests in the repo assume specific architectural options (specifically: MTVEC is completely writable, and that HW misalign is not supported) and that implementations that fail tests that assume the opposite should apply for a waiver when applying for the trademark.

Ubilite, 2020-12-10:
Hi Allen,

I understand that the tests appear to work for the majority of implementations but would it be problematic if we can add 2 or 3 lines to skip/pass the test after attempting to write "C" to 0 and issuing a read and seeing that the write failed? This would at least fix the C extension problem with misa (the WARL register practically present in all implementations) without waiting for riscof. Especially, considering that an attempt to disable the C extension is already there in the code.

Concerning the MTVEC problem, I understand that is more complex and no need to kill the test for it.

allenjbaum, 2020-12-11:
This will happen automatically when framework v2.0 gets merged. It will
have dynamically generated reference signatures (from the formal model).
We've been hesitant to divert what little resources we have for a short
term fix when a longer term fix is coming.
In the short term, the tests have been completely rewritten and will be
merged shortly- though that can't fix the issue

The root of the problem is that, depending on a "hidden" configuration
(misalign), we can have two different reference signatures, and the old
framework only knows how to deal with one.
So we can make one assumption and some cores will break, or the other, and
other cores will break - or not test it at all, which I think is a bad
compromise.
I don't think we will get to that fix this year... but the year is nearly
gone.

eroom1966, 2020-12-11:
@allenjbaum 
Hi Allen
Do you have a list of the ‘majority of implementations’ that this is working ?
The only implementations I have seen added to the repo are ibex, cv32e40p, and ri5cy
Thx
Lee

allenjbaum, 2020-12-12:
I'll have to confess that this is because of the lack of evidence of anyone
else requesting changes.
 I do know of at least 2 custom models out there that work with these
existing tests, and will break if we make changes (that is, default from
misaligned support vs not)
I am also not comfortable with dropping tests completely and losing
coverage - which is my interpretation of the request.

This is a basic shortcoming of the existing framework and a major reason
for moving to the riscof framework ASAP.
Misaligned support is far nastier than yes or no, unfortunately.
There is no requirement for an implementation to consistently trap or not
trap on a misaligned access - even to the same address.

In any case, I'm not going to risk short term fixes that might break tests
for some (unknown number of) custom models to fix tests for these when we
have a long term fix coming.
Let me see if there is some other way to restructure tests so avoid this
particular issue

In the meantime - if someone needs to ask for a waiver because some
particular test fails, and they can blame it on us, I'll take the blame and
approve a waiver.

allenjbaum, 2022-08-11:
This is fixed with new trap handler; will be closed when that is merged.

pawks, 2023-02-09:
Addressed by PR #277

closed 2023-02-09 by pawks