riscv/riscv-arch-test #149: Support for OpenHW/CV32E40P target implementation

Imperas, 2020-12-02:
I have outlined 4 issues related to tests that are deficient in their implementation
https://github.com/riscv/riscv-compliance/issues/145 https://github.com/riscv/riscv-compliance/issues/146 https://github.com/riscv/riscv-compliance/issues/147 https://github.com/riscv/riscv-compliance/issues/148

None of these tests will pass in their present form on either ibex or OpenHW/CV32E40P
In the case of CV32E40P, we are approaching RTL freeze and believe that there are no known issues in terms of meeting compliance, except for those stated in the four issues above.
In which case I believe it would be prudent to deprecate those tests until they are improved to handle the possible design choices the RTL has taken from the RISCV specification.
For these issues outlined I think the tests need to be enhanced for the following categories

- Ability to describe an MTVEC register which provides alignment to address boundaries greater than 2 byte alignment
- Ability to describe a target has support for the C extension
- Ability to describe a target has Hardware support for misaligned load/store accesses
- Ability to describe a target which implements DEBUG mode

pdonahue-ventana, 2020-12-02:
> Ability to describe a target which implements DEBUG mode

The existence of debug mode is completely invisible to software running on the hart.  The only ways to enter debug mode are via:
- haltreq or resethaltreq requests from the debug module (the DM is only accessible from the external debugger via some sort of debug transport module such as JTAG)
- triggers in the trigger module which have action=1.  That requires dmode=1 which can only be written by the external debugger.
- ebreak instruction when one or more dcsr.ebreak bits are set.  The dcsr.ebreak bits can only be written by an external debugger.
- single-stepping via dcsr.step which can only be written by an external debugger.
- the hart is part of a halt group at the request of the external debugger

The ability to describe a target which implements debug mode is important for testing debug mode and external debug capabilities but not for testing the unprivileged and privileged architectures or even for testing the non-external portions of the debug architecture such as trap-based debug (action=0).

allenjbaum, 2020-12-03:
The other approach, if time is short for ratification, is to ask for a waiver for failures on those tests.
The RFQ tests were specifically designed for a single HART configuration (e.g. alignment was supported or not), so we would have to modify something for this

allenjbaum, 2020-12-03:
The consensus is that we will add a note into the README that indicates that the tests in the repo assume specific architectural options (specifically: MTVEC is completely writable, and that HW misalign is not supported) and that implementations that fail tests that assume the opposite should apply for a waiver when applying for the trademark.

allenjbaum, 2022-07-12:
OpenHardware has ported their environment to riscof (or vice-versa).
So: it is now time to confirm that you're no longer seeing these issues. Or that you are.
Note that in addition to installing riscof, you'll also need to construct the YAML fill that describes:
whether misalign is supported or not, which bits of MTVEC are writable, whether C extension is supported  (I'm puzzled over this one, since I thought that was supported in the old framework),

Imperas, 2022-07-12:
Allen - the core/verification environment we reported this issue for was the CV32E40P - which was RTL frozen/signed off over 18 months ago and so we got the compliance waiver for that then - this core verification is unlikely to be run again for compliance purposes by OpenHW with riscof... of course being open source - you could clone the repo and do this.

There are currently 8+ other cores/core variants in various stages of verification within OpenHW group - when you say 'OpenHW has ported their environment to riscof' can you explain where this information comes from and which core you are referring to.

thx,
Simon

allenjbaum, 2022-07-13:
this is the OpenHardware reference:
https://github.com/openhwgroup/core-v-verif/pull/1254.

If the CV32E40P is not going to self-certify, then could you close issues
146-149 here:
https://github.com/riscv-non-isa/riscv-arch-test/issues

Imperas, 2022-07-14:
@allenjbaum Your comment "OpenHardware has ported their environment to riscof" I think is not correct - what the CVA6 group in OPenHW are doing is using the tests in the riscv-arch-test github - I cannot find any use of the riscof python framework - just the tests are used - I will keep looking.
Simon

allenjbaum, 2022-07-14:
Hmm; you might be correct there; the existing support was bit-rotted, and
if it now works on the current tests, it should work with riscof,
because the tests haven't changed, nor has the interface to the DUT; its
primarily how the model is called. But: given the presentation today, I'm
pretty sure that there will be configuration problems, but fewer than what
Stacy had.

Do you have opinions about microcontroller implementations that don't have
enough memory to properly test the address ranges?

allenjbaum, 2022-08-11:
We believe the root of this issue is that MTVEC wasn't completely writable. This has been addressed in the new trap handler and will be closed when that is merged.

allenjbaum, 2023-02-22:
The trap handler has merged, so closing this issue. If this is demonstrated to not work, reopen it

closed 2023-02-22 by allenjbaum