riscv/riscv-arch-test #156: PMP/PMA test

panda1628, 2020-12-08:
Is there any PMP/PMA compliance test?

allenjbaum, 2020-12-08:
no, not yet. There are some older tests in the riscv-test directory, but
those aren't official, and I don't think coverage is up to what we intend
to test.

allenjbaum, 2020-12-08:
Oh, and PMAs are very specific to every implementation, so it isn't clear
how we could possibly test them.

panda1628, 2020-12-08:
Can you please give me the path? 
"There are some older tests in the riscv-test directory"

I am looking for an example of PMP/PMA test, thanks

pdonahue-ventana, 2020-12-08:
https://github.com/riscv/riscv-tests/blob/master/benchmarks/pmp/pmp.c

allenjbaum, 2020-12-22:
Note that these older tests are self checking, and don't conform to the test format spec, so can't be run with the architectural test framework.

allenjbaum, 2021-04-22:
question answered, I'm closing this issue

closed 2021-04-22 by allenjbaum