riscv/riscv-arch-test #160: 64-bit Misaligne load/store Privilege Tests

duncangraham-Imperas, 2020-12-21:
I am trying to execute the 64-bit privilege tests but they are failing the signature verify. I have attempted to debug the failure and it appears that the expected mtval is not correct. This causes differences in the signature, in the example shown below the expected is zero and the signature contains a non-zero value
```
Check           misalign-ld-0121,22c21,22
< 00000000
< 00000000
---
> ffffffb9
> ffffffff
29,30c29,30
< 00000000
< 00000000
---
> ffffffba
> ffffffff

```etc.

I have executed using the riscv-isa-sim HEAD (Spike RISC-V ISA Simulator 1.0.1-dev) and also using sail which produce the same failure.
What is wrong here?
I can see no requirements to run with a specific version of either of these simulators or any version information about the version used to verify before checkin.

neelgala, 2020-12-21:
I just ran the whole suite 64/32 with the latest version of [spike](https://github.com/riscv/riscv-isa-sim) which was updated less than 2 days ago and all tests seem to be passing at my end. Not able to find branch/tag : 1.0.1-dev

Is there a possibility that you could share the elf of the test that you are generating and running on spike ?

duncangraham-Imperas, 2020-12-21:
It looks like I was working with the version just before these changes were applied 2 days ago, so I will pull these changes and run again. Will let you know the result.

duncangraham-Imperas, 2020-12-21:
So, I was using the version prior to the last update (2 days ago) and this was failing but updating to the latest change and I am seeing all tests passing.

closed 2020-12-21 by duncangraham-Imperas