riscv/riscv-arch-test #165: Version numbering

towoe, 2021-01-14:
I am a little bit confused by the tags used for the releases.
I see the tags v0.1, v1.0, v2.0 and v2.1.
I would assume all prior to v2 is based on the "old" riscv-compliance framework and the starting with v2 it is based on RISCOF?

But I see some other version numbering referring to v0.1 and v0.2 only.
For example the [RISC-V specification](https://riscv.org/technical/specifications/) page.
And the [migration guite](https://github.com/riscv/riscv-compliance/blob/master/doc/MIGRATION.adoc#1-migration-from-v01-to-v02).

Maybe someone can explain/clarify the meaning of the versioning? Thanks

neelgala, 2021-01-14:
i think all v0.2 instances should become v2.0 since thats what the tags say.

Also v2.0 is not riscof based yet. v2.0 is new set of upgraded tests with slight framework changes from the previous versions. The change to RISCOF will be v3.0 or higher. 

Thanks for the poiners on the discrepancies.. i'll create a PR for these fixes.

towoe, 2021-01-14:
> i think all v0.2 instances should become v2.0 since thats what the tags say.

Thanks for the clarification!
> Also v2.0 is not riscof based yet. v2.0 is new set of upgraded tests with slight framework changes from the previous versions. The change to RISCOF will be v3.0 or higher.

Might be good to put that into the documentation as well.

allenjbaum, 2021-01-14:
I have a list of features that will be added to the erepo prior to v3.0,
which will be riscof.
These will all be accompanied by minor  number changes, e.g.from 2.1 ->2.2.
Adding tests should probably be 3rd digit changs, e.g. if we added FP tests
right now, I'd probably bump v2.1 to v2.1.1.
Documentation updates will be similar.
The changes I've identified that go beyond that are: (in no particular
order)
 - removal of target directories from the repo (to be replaced with
documentation s to where to find the target models and how to use them with
the framework
   This will  probably be done gradually, and I'm not sure I want to change
the minor version for each one of those after the first
 - out-of-line assertion changes, that reduce the size of code when
assertion reporting is enabled, and simplify model specific macros
 - assertion value generation using the reference model
 - non-determinism support (probably after riscof)

InspireSemi, 2021-01-14:
V1,0 was the Tag applied to just before the new "upgraded tests" was added. 
v2.0 was the Tag just after the new "upgraded tests" were added.
v2.1 was after a minor change the the new "upgraded tests"

Plan is for v2.x to be only for the "upgraded tests" 
We will have to determine if just one number 2.x for a sub version is ok or how far down the rabbit hole we would like to go.
v2.x.y.z?    

V3.0 will be riscof when it is deployed.

allenjbaum, 2021-01-15:
2.x.y: the y component should only be updated for clarifications, minor
grammar/spelling, etc.
I'm sure someone has some standard definition of the conditions for
incrementing each component of the version.

InspireSemi, 2021-01-15:
There is no standard.  It is whatever you want it to be.  I've seen version strings that are 20+ characters that include the date and other info.   I think 3 numbers is more than enough a major minor and sub minor number..

ben-marshall, 2021-01-15:
There actually is a standard for communicating types and severity of changes through version numbers. It would be great if this or something like it could be adopted. It looks like it already has been in part.

Semantic Versioning: https://semver.org/

The summary is the only bit you really need to read, the rest is rather heavyweight:

> Given a version number MAJOR.MINOR.PATCH, increment the:
> -   MAJOR version when you make incompatible API changes,
> -   MINOR version when you add functionality in a backwards compatible manner, and
> -   PATCH version when you make backwards compatible bug fixes.
> 
> Additional labels for pre-release and build metadata are available as extensions to the MAJOR.MINOR.PATCH format.

neelgala, 2021-01-15:
we use semantic versioning in RISCOF as well.

MarcKarasek, 2021-01-15:
If you want to call this a standard ok..  It is just one man's idea of how
to do things..

I have seen versioning in eerie kind of form possible..

His way of doing it is as good as any other.. We have not really followed
it as 2.0 should have really been 2.0.0

allenjbaum, 2021-01-16:
I guess I'lll have to ask: does semantic versioning require trailing zeroes?
Probably, though v2.1 should have been 2.0.1, since it really was a bug fix
(you might be able to argue it the other way.
I like the semantic versioning definition that Ben pointed at; let's use it.

allenjbaum, 2021-04-22:
I think this has been discussed, beat to death, and implemented, so I'm closing this

closed 2021-04-22 by allenjbaum