riscv/riscv-arch-test #169: RISCOF redefine of `TEST_CASE_1`

towoe, 2021-01-22:
https://github.com/riscv/riscv-compliance/blob/65cd9556846c60608c06e9db22ab573c1e86fe6b/riscv-test-suite/env/arch_test.h#L14

If I use the test suite with RISCOF `TEST_CASE_1` will be defined twice, once from `arch_test.h` and again with the compiler invocation (test file: `RVTEST_CASE`, database.yaml, plugin (testentry['macros'])).

When I create a test suite with riscv_ctg, `arch_test.h` does not contain this definition. But I am not sure if this is needed for the current checks in riscv-compliance. Maybe @neelgala or @pawks can help to clarify this?
The file also differs slightly in two other places (`adj_mtval`, `sv_mtval`).

Thanks

pawks, 2021-01-22:
The macro `TEST_CASE_1` is defined in riscof as a part of command line arguments to enable partial test selections in scenarios where its necessary. This contributes to the dynamic test and test case selection based on the configuration of the DUT. The current framework over here doesn't have the capabilities to handle dynamic test case handling, which is why the `TEST_CASE_1` macro is defined by default in the header file.

towoe, 2021-01-22:
Thanks for the clarification @pawks.
So I guess this will not change until v3 is ready and just be how it is? (Just to make sure)

allenjbaum, 2021-04-22:
Yes, its there so when v3 is deployed, no changes to tests will be necessary. 
I believe this is now answered (very belatedly), so I'm c;osing this.

closed 2021-04-22 by allenjbaum