riscv/riscv-arch-test #170: Privilege mode test

panda1628, 2021-02-19:
I am looking for privilege mode tests (e.g. machine, supervisor and user)
Switching between different modes and access different privilege CSRs cause exceptions

Are they WIP?

allenjbaum, 2021-02-20:
The work on those is just being started.
Three are some older tests, with minimal coverage in the riscv-tests
directory, but the are pretty basic and are incompatible with the current
framework.

panda1628, 2021-02-20:
Thank you for the update. 
Let me know if I can help

allenjbaum, 2021-02-24:
Bit of a late reply - I've been trying to figure out how you can help.
Without knowing your background, it is hard to say.
First: you probably should join the sig-arch-test mailing list here:
https://lists.riscv.org/g/sig-arch-test/

The next major project is to transition to framework 3.0.
That framework will
 - take a YAML format description of someone's implementation,
 - configure the Sail formal model to match,
 - run architectural tests with that configuration, generating a signature,
 - run it on the user's model, generating another signature
 - compare the signatures and generating a test report.
That test report will include a test pass/fail report, as well as details
of what is being tested, etc.
That report can be added to the github.com/riscv/reisc-test-reports repo to
demonstrate /self-certify architectural compatibility.

The biggest prerequisite to getting this done is currently retrofitting the
Sail model to e able to configure itself.
We are hoping that RIOS Labs - a group contributor" may take on that work.

The next prerequisite is a test-plan for the privileged architecture: what
are we going to generate tests for, what are the cover points that we need
to hit to show that we've tested enough.
That results in a YAML format file that is used to measure the quality of
the tests, and discover if we've missed any corner cases.

Finally, we have to generate tests. This should be done by writing test
generators - probably for each priv feature on a section by section basis.
In theory, that will be done by another group contributor. (also RIOS Labs?)
--------
We are starting up a Simulator SIG, and are looking for a chair to drive
it. The role of that SIG is to manage / guide the development of our
simulator technology by these group contributors.
That includes Sail, Spike and QEMU - and possibly GEM for some cases. We
have a vice-chair nominatoed.

Beyond that, there is the management of whoever is writing tests: reviewing
tests plans and managing/guiding development of tests.
--------
On the arch-test-sig side, we need to maintain the existing framework, and
also evolve it to handle non-deterministic results, memory mode/coherency,
and asynchronous events.
Some of that will require an entirely different approach than what we are
doing now, so that will need to be architected
--------

I'm also nominally in charge of
 - a Continuous Integration/Tests group to make sure that any new tests and
SW gets automatic regression testing if anything changes
 - a DevOps group, doing....DevOps stuff
 - a Documentation Group to establish tools and standards for all the docs
that will need to be produced.
You begin to see why this reply has taken so long...I've been focusing
mostly on the Architectural Tests for a while, ad need to concentrate on
Simulators, now as that is the blockage.

With all the work list above: do you see anything that you'd like to help
out with?

allenjbaum, 2021-04-22:
question answered, closing this issue

closed 2021-04-22 by allenjbaum