riscv/riscv-arch-test #171: “Zifencei” Instruction-Fetch Fence, Version 2.0

panda1628, 2021-02-24:
According to “Zifencei” Instruction-Fetch Fence, Version 2.0:
"The FENCE.I instruction was previously part of the base I instruction set. Two main issues are driving moving this out of the mandatory base, although at time of writing it is still the only standard method for maintaining instruction-fetch coherence."

I-FENCE.I-01.bfd test shall take illegal instruction exception withOUT this option, correct?

pdonahue-ventana, 2021-02-24:
Section 2.2 says: "The behavior upon decoding a reserved instruction is UNSPECIFIED."  So it may take an illegal instruction exception or it may not.

panda1628, 2021-02-24:
Can we configure the test accordingly?

allenjbaum, 2021-02-24:
The Zfencei op is only defined for a hart that implements the *optional*
Zifencei extension correct?

If the DUT doesn't implement that extension, it shouldn't be executing that
test.

Where is this I-FENCE.I-01.bfd test located?
The current framework has only this test:
riscv-arch-test/riscv-test-suite/rv32i_m/Zifencei/src/Fencei.S
- not sure where the .bfd extension comes from

If you're running the test suite on a model that doesn't implement it, you
need to either change the makefile to avoid it, or perhaps delete the
Zifencei subdirectory

panda1628, 2021-02-24:
rv32Zifencei/src/I-FENCE.I-01.S  is the test
Sure, I will NOT run it 

My suggestion: Compliance tests can be configurable according to the target implementation
Input: The base + options 
Output: A set of compliance tests for the target

allenjbaum, 2021-02-24:
I don't know which version of the repository you are using, but it is
certainly out of date at this point.

Your suggestion is precisely what Framework 3.0 does.
But there is work to do to make that actually usable, as I've outlined.
The number of possible options and behaviors is staggering.

allenjbaum, 2021-03-20:
You are not using the current repo or test located in 
    https://github.com/riscv/riscv-arch-test/riscv-test-suite/rv64i_m/Zifencei/src/Fencei.S
so I'l be closing this

closed 2021-03-20 by allenjbaum