riscv/riscv-arch-test #178: loss of test spec - pls bring back

Imperas, 2021-03-29:
I see a note on spec/TestFormatSpec.pdf is removed since its old. Keeping only adoc file

this is wrong - who approved this pull request??

the spec is the spec of the 0.2 framework and it is set in stone (if it was a ISA spec it would be ratified) so it must have 'approved' status - so it can't be changed until a new spec is 'approved' - and that will be 0.3 with a selection framework. Any edits to the adoc are just a work in progress and do not change the spec as that is fixed.

the file as a pdf needs keeping in the repo as the 'approved' spec of  0.2 

pls put it back into the Spec dir and TestFormatSpec.pdf with an appropriate date and version number [Allen will have the original 'approved' one] as everyone relies on it and it cannot be changed - so a pdf is good.

and please make the adoc one as a draft of the next spec.

I have attached my copy of the testformat spec.

thanks
[TestFormatSpec.26.feb.2020.pdf](https://github.com/riscv/riscv-arch-test/files/6223324/TestFormatSpec.26.feb.2020.pdf)

neelgala, 2021-03-30:
This was approved in #176 .

I am not sure what is "0.2 framework" can't seem to find it anywhere. Maybe you meant 1.0? 

If so, then [that tag](https://github.com/riscv/riscv-arch-test/tree/1.0) already exists and you can checkout that tag which will present the same doc along with the code which matches the doc. This is the same what other RISC-V specs are following - keeping a copy of the latest draft (in adoc or pdf format) while tags/realeases preserve the older versions.

Removing an outdated doc which matches nothing to the current code base, seemed like a good thing.
If @MarcKarasek @allenjbaum feel otherwise, I can put that file back in.

Imperas, 2021-03-30:
OK. - I think you need a pdf in tagged repos/releases. The one in that tag, [TestFormatSpec v1.2.5 ](https://github.com/riscv/riscv-arch-test/blob/1.0/spec/TestFormatSpec.pdf)is the only one that was put out to review and used as approved.

So can OpenHW (or other users) go back to the 1.0 framework using "that tag (https://github.com/riscv/riscv-arch-test/tree/1.0) " and run those tests and request approval as a compliant compliance 1.0 core  - (it will take  2-3 days to port to the 2.0 framework (current) and I don't see the appetite)  and the idea of tagged releases is they are a complete consistent set - and that set works - the repo, the pdf, the tests all match with what people were using until recently and now you change things - so when you tag a release users can claim compliance to that release? is that the idea - or will users have to claim 'compliance on this day version of the repo'? or will they constantly need to keep updating their environments on a daily basis as you make changes to the arch test framework, test, extensions, specs and repo?

[Or will users only be able to claim compliance for their core when there are arch tests for all (yes ALL) the specs they have implemented - for example a RV64GC core can only be compliant when there are full arch validation tests in this github for I,M,C,A,F,D, Supervisor, Machine, User ie all the current ratified (2019)  specs???]

To me - you made a release, 1.0 and so users should be able to use that to run against that release and thus claim compliance on success,
and when you move to 2.0 or 2.1 etc... they could choose to test against that too - in fact - to any version you put a release tag on.

So in fact users should only ever user release tagged versions (much like ISA specs) [and hence my comment about the pdf should reflect the release level] - and now you have enlightened me - it is the whole repo that needs 'setting' with a release tag.

This means the repo would then just be a work in progress, often would be broken (with new tests, incomplete spike/sail for new ISA subsets, etc.) - which is all fine as a work in progress - and the release versions to be tagged would always be very carefully reviewed as you near version release (probably coinciding with spec extension ratification) , and would be high quality and the thing the majority of users would utilize.

I think the process of repo releases, quality assurance testing the repo, tagging releases, ratified tests, unratified specs, stability, progress, random changes and new frameworks warrants discussion and a process needs defining and agreeing/adopting... [Then we wont have issues like we had before of adding CSR values to signatures which broke the tests - just being added the day before a major update and review... and no regression tests failing]

Thanks. Simon

allenjbaum, 2021-04-21:
The .pdf file is now located in the release area: https://github.com/riscv/riscv-arch-test/releases/tag/x.y.z
where 2.4.1 is the most recent x.y.z.

closed 2021-04-21 by allenjbaum