riscv/riscv-arch-test #179: Why does .gitignore ignore riscv-targets/?

ben-marshall, 2021-03-30:
Hi there
I was just experimenting with augmenting the Sail targets to handle the new scalar crypto tests and noticed that `.gitignore` contains this line:

https://github.com/riscv/riscv-arch-test/blob/da00add4951dc5e1371d8478cf3a46bfc8c9456a/.gitignore#L11

Is there a reason for this? I saw it was introduced with the RFQ tests, but couldn't see why. It makes altering existing targets a bit tricky.

Thanks,
Ben

neelgala, 2021-03-30:
The riscv-target folder will typically contain target-specific files/information which should not be checked into the arch-test repo - hence the .gitignore. This was somewhat of a requirement from users, to prevent accidental check-ins of confidential target-files.

This will however prevent the git diff to work on the files under riscv-target (if that's what you are doing). You could always rename the folder to workaround it.

Also soon the riscv-target folder will be discarded altogether and replaced with readmes pointing to the SAIL and spike arch-test/target readme in their respective repos. Arch-test repo would no longer be hosting target specific files. Waiting for @allenjbaum to signal when this can happen.

ben-marshall, 2021-03-30:
Hi @neelgala 

Thanks for explaining, that all make sense. I saw that had already been done for Spike, (hence me putting in riscv/riscv-isa-sim#687) so I'll keep all my modifications local for now.

Feel free to close this, or leave it open as a note to others until things are moved around.

Cheers,
Ben

allenjbaum, 2021-03-30:
Too many things on my plate, but it is on my perpetual list of things to do
- not at the top, though.

closed 2021-04-21 by allenjbaum