riscv/riscv-arch-test #180: Proposal to reduce memory requirement below 64KByte

bobbl, 2021-03-30:
I am using the tests to check my SoftCPU on a small FPGA board with very limited memory. And there are many similar CPU projects with small memories that use the compliance tests, because the coverage is good and the binaries used to be small. However, with the new framework, the JAL test requires 1.7 MB of memory and the branch tests require 300KB. That is too much for small embedded CPUs without MMU. Issue #157 addressed this problem, too.

Here is a proposal, how to get the memory requirement below 64KByte:

 1. Split `jal-01.S` into two files, one with small offsets and one with large offsets. The latter contains only 3 test cases, but they cause 1.7 MB of the file size. Without them, the other test cases require only 4K. A CPU with small memory can then state that it passes all tests, except the test for large JAL offsets.
 2. The branch tests also contain large runs of nops to implement the distance between instruction and branch target. By re-arranging them, the spaces can be shared between multiple cases.
 3. All other tests are below 64K

As proof of concept, I converted the 6 source files of the branch tests for RV32 with the help of an awk script. The scripts and results can be found in [my fork at riscv-test-suite/rv32i_m/I/](https://github.com/bobbl/riscv-arch-test/tree/master/riscv-test-suite/rv32i_m/I). What I did:
  * All cases with offset N < 0x100 are not modified
  * Cases with larger offsets are grouped together. There are only four groups with offsets 0x100, 0x200, 0x400 and 0x556.
  * Every group with offset N consists of four blocks:
      - Block 1: N/2 forward jumps with relative offset  N + 12
      - Block 2: The backward test cases with the appropriate offset. They jump to block 1 and from there back to this block, but at the position where the correct signature is written.
      - Block 3: The forward test cases, jumping into block 4 and back to block 3 where the signature is written
      - Block 4: N/2 backward jumps with relative offset -N + 12

Special cases:
  * Additional jump blocks are necessary, when the accumulated code size of  the tests is larger than the offset
  * Some tests use different signature pointers/registers. Care must be taken to write to the correct position within the signature 

To implement the newly arranged branches I added these macros to `arch_test.h`:
   
    #define TEST_BRANCH_BWD_RETURNBLOCK(imm)         \
        j . + (2*imm)                               ;\
        .rept (imm-2)/2                             ;\
            j . + (2*imm) + 12                      ;\
        .endr
    
    #define TEST_BRANCH_BWD_OP(inst, tempreg, reg1, reg2, val1, val2, imm, swreg, offset, sigtag) \
        LI(reg1, MASK_XLEN(val1))                   ;\
        LI(reg2, MASK_XLEN(val2))                   ;\
        addi tempreg,x0,0                           ;\
                                                    ;\
        inst reg1, reg2, . - (2*imm)                ;\
        addi tempreg, tempreg,0x2                   ;\
        j 1f                                        ;\
        /* taken branch returns here from block 1 */ ;\
        addi tempreg,tempreg, 0x1                   ;\
                                                    ;\
    1:  RVTEST_SIGBASE(swreg,sigtag)                ;\
        RVTEST_SIGUPD(swreg,tempreg,offset)
    
    #define TEST_BRANCH_FWD_OP(inst, tempreg, reg1, reg2, val1, val2, imm, swreg, offset, sigtag) \
        LI(reg1, MASK_XLEN(val1))                   ;\
        LI(reg2, MASK_XLEN(val2))                   ;\
        addi tempreg,x0,0                           ;\
                                                    ;\
        inst reg1, reg2, . + (2*imm)                ;\
        addi tempreg, tempreg,0x2                   ;\
        j 1f                                        ;\
        /* taken branch returns here from block 4 */ ;\
        addi tempreg, tempreg,0x3                   ;\
                                                    ;\
    1:  RVTEST_SIGBASE(swreg,sigtag)                ;\
        RVTEST_SIGUPD(swreg,tempreg,offset)

    #define TEST_BRANCH_FWD_RETURNBLOCK(imm)         \
        j . + (2*imm)                               ;\
        .rept (imm-2)/2                             ;\
            j . - (2*imm) + 12                      ;\
        .endr

Maybe the compliance test generator riscv-ctg can be modified to arrange the tests in this way to reduce the memory requirement from 300K to 64K.

allenjbaum, 2021-03-30:
We've discussed doing this. And you aren't the first person to have run
into this: https://github.com/riscv/riscv-arch-test/issues/157
Testing that jump offsets can be correctly calculated means you have to
test that the MSB(-1)  of the immed field can be set (for forward branches)
or cleared (for backwards branches).

What we've documented is that it is up to the implementation model to
provide sufficient resources to run the tests.
That means you need to pretend that the memory outside of the
implementation of your hart is big enough.
You can build a chip  with less memory, and say it is "RISC-V compatible"
(if you meet all the branding conditions), but you must do at least that.

allenjbaum, 2021-04-21:
This has been answered, and is a duplicate, so closing it.

closed 2021-04-21 by allenjbaum