riscv/riscv-arch-test #181: Getting NOP/HINT Instructions in Compressed Category tests. Are they really needed?

animesh-misra, 2021-04-03:
Hi,

In Compressed category, some tests are having HINT Instructions which like the NOP instruction do not change any architecturally visible state, except for advancing the pc and any applicable performance counters.

e.g. - caddi-01 Test [32 bit]

https://github.com/riscv/riscv-arch-test/blob/da00add4951dc5e1371d8478cf3a46bfc8c9456a/riscv-test-suite/rv32i_m/C/src/caddi-01.S#L62

[image]
_Figure 1:Showing line 62 where, Instruction 5 ->" c.addi x10,0x0"_  

Here, the compressed addi instruction is having an immediate data value of 0x0.

[image]
_Text from RISC-V spec: Volume 1._

As stated in Spec, Immediate data value should not be 0x0 otherwise it is encoded as HINTs. Similarly, there are other tests in the Compressed category which are containing these HINT Instructions.

Is there any reason these instructions are present in the tests?

aswaterman, 2021-04-04:
This is a good test to have.  It's important to test that HINT instructions don't change architectural state.  This test does so in a limited way: it ensures that x10 is not changed by executing a HINT whose rd field is x10.

animesh-misra, 2021-04-05:
Alright.

Imperas, 2021-04-05:
What?  - this test is not a test for HINTs it is a test for compressed addi.
If we want to test HINT instructions - that is another class of test and requires a whole load of new tests.
this instance was just a mistake in the test data values where it actually did not do what the test writer intended.
this test should be fixed and should not use reserved instructions.

Imperas, 2021-04-05:
pls re-open until properly resolved.

neelgala, 2021-04-05:
There are multiple hint instructions in compressed spec. All those instructions have been carefully placed in the respective instruction tests rather than having a dedicated test for hints. The particular test-case is not a reserved.. its a hint and thus a valid test-case as anderw pointed out.

eroom1966, 2021-04-05:
@aswaterman 
> This is a good test to have. It's important to test that HINT instructions don't change architectural state. This test does so in a limited way: it ensures that x10 is not changed by executing a HINT whose rd field is x10.

I would tend to agree with you Andrew, what you are describing would be a case of NEGATIVE testing, in fact in the early days of the compliance group I argued for including tests that assured behavior which should NOT happen be included in the architectural tests. The decision was taken that negative tests would NOT be included, and only positive tests would be run on a target.

@neelgala 

> The particular test-case is not a reserved

in the testcase above, I see the following example
c.addi x10,0x0
According to the manual, section 16.7 HINT Instructions, table 16.7 this instruction encoding is defined explicitly as

> **Reserved** for future standard use

and Reserved instructions (from section 2.2) 

> The behavior upon decoding a reserved instruction is UNSPECIFIED.
> Some platforms may require that opcodes reserved for standard use raise an illegal-instruction
> exception. Other platforms may permit reserved opcode space be used for non-conforming extensions.

In which case, if these instruction encodings are indeed reserved, and the behavior of reserved is defined as UNSPECIFIED, how can you determine the valid signature to check against ?

Maybe this is ambiguous in the manual from the actual intent of this HINT instruction ?

Thx
Lee

allenjbaum, 2021-04-05:
Is this just a disagreement about in which .S file a particular hint should
be tested?
There is no*technical* reason why it couldn't be done one way or the other
The reasons it should be done one way or the other are ease of managing the
test(s) and ease of generating them

I would agree that this is a very simple test; a true hint test would
actually check more than just GPR  state,
  - it would test all architectural state in the hart - which primarily
includes all CSRs.
I could see another class of test to do that, but I might put that into the
class of tests which are much more DV focused.

If we added that level of testing, a single test (or test directory) for
hints could amortize the cost of checking all CSR state.
That might extend to amortizing test generation methodology as well.

But that can't be done with this v2 of the framework; until we get to
riscof, there is no way to even encode in the tests which CSRs exist .

With separate tests, when an extension is added, only the new extension .S
files are changed (assuming they add hints).
With combined tests, we'd have to update the hint tests as well; probably a
minor management issue

eroom1966, 2021-04-05:
@allenjbaum 
Hi Allen
our postings almost overlapped, I think there are 2 questions here

1. Is it the intent to check that architectural state is not modified by executing instructions which are not architecturally visible ? I would argue that this is a form of negative testing, you argued strongly against negative testing when I (and a number of others) was in favor of it.

2. Is this HINT instruction decoding reserved ? (it would appear it is) in which case should instructions be tested with UNSPECIFIED behavior ?

Thx
Lee

allenjbaum, 2021-04-05:
Hmm, 2021.03.25 unpriv spec section 17.5 under c.addi says explicitly:

The code points with rd=x0 encode the C.NOP instruction; the remaining code
points with nzimm=0 encode HINTs.

Under 17.7 (the old 16.7) it says:

RVC HINTs are encoded as computational instructions that do not modify the
architectural state,
either because rd=x0 (e.g. C.ADD x0, t0), or because rd is overwritten with
a copy of itself (e.g. C.ADDI t0, 0)

Clearly, this should include cases where the immediate value is 0 for some
instructions at least.
It also says explicitly in table 17.3

Reserved for future standard use

I think context here is important; these hints as being reserved for
standard use, as opposed to the following instructions which are

Designated for custom use

So perhaps the fix here is to change the word "reserved" to "designated" in
the top part of the table.

neelgala, 2021-04-06:
@eroom1966 

The table you are refering to talks about reserved/custom HINTS  which is different from reserved/custom instructions. Whether a HINT is reserved or custom, the fact that it shouldn't change arch state always holds. And the test-case in question does exactly that (not an exhaustive check of all arch-state as that would lean over to DV).

allenjbaum, 2021-04-06:
Exactly
      It is   reserved   for *standard* use vs.   reserved   for *custom* use.
Or, it is designated for *standard* use vs. designated for *custom* use.

So, it is talking about hints which only RVI can define vs. those that anyone can define for their own use.
It has nothing to do with a reserved opcode that may be undefined.
They are noop hints in all cases, perfectly legal to use, and should cause no change to architectural state.

I have asked Krste to have that single word "reserved" changed to "designated" which should should down this entire ridiculous discussion. 

This issue should be closed and stay closed whether that word is changed or not.

closed 2021-04-06 by allenjbaum