riscv/riscv-arch-test #186: privilege tests fail

vskalagi, 2021-04-20:
I am using privilege tests to check my custom RISC simulator with support to I, M, and F extensions. But it is failing with 8/16 tests. So, I also tried to check these tests with spike simulator by using --isa=rv32imf. Even for the spike simulator, these tests were failing. But when I changed --isa to rv32ic(which was default), there are no failed tests. Do these tests require targets that support the 'C' extension?

neelgala, 2021-04-20:
Can you list out the tests that actually failed. I would suggest reading the disclaimers [here](https://github.com/riscv/riscv-arch-test#test-disclaimers). The disclaimers indicate under which scenarios waivers can be used.

vskalagi, 2021-04-20:
These are the tests failed:
misalign2-jalr-01
misalign-beq-01
misalign-bge-01
misalign-bgeu-01
misalign-blt-01
misalign-bltu-01
misalign-bne-01
misalign-jal-01
ebreak

neelgala, 2021-04-20:
As in point-2 of the disclaimers: 

> Targets without the compressed extension support will fail the following tests: rv[32/64]i_m/privilege/src/misalign-b[ge[u],lt[u],eq,ne]-01.S

So yes, if you don't have compressed extension support these test fails are okay to have. The next version of the framework - RISCOF - has the capibility to configure the test (and therefore signature) to work for both compressed and non-compressed targets.

I hope this answers your question, if so kindly close the issue.

closed 2021-04-20 by vskalagi

MarcKarasek, 2021-04-20:
Which specific tests failed?

Can you send the output from the tests

Imperas, 2021-04-20:
If the tests need C support, then they should be in the C directory?

allenjbaum, 2021-04-20:
This issue is actually a duplicate of issue#22 in the repo.
This was a deliberate choice we made, because those tests should be run
regardless if C-extension is implemented or not - they merely give
different results.
The current framework only allows a single static reference signature, as
I'm sure you know.

So we have a choice of  a) not testing at all, or b)  testing and failing.
We chose b) because a) risked a false negative: a bug we should have caught
but didn't, which is worse than a false positive which we can look at and
waiver.
In addition, by choosing the latter case, when we move to riscof, the tests
don't need to be re-written or moved, they'll just work.
There are several other issues in the repo that get fixed when we move to
riscof.