riscv/riscv-arch-test #189: proposal to enhance the RVTEST_ISA macro

neelgala, 2021-04-27:
For the purpose of explaining the issue, consider the ROL instruction which is currently used by both "K" and "B" extensions. Which means we could use the same test for implementations that implement either of the extensions (I know there other subsets of Z* which can enable this, but lets keep it simple for this example). In which case the RVTEST_CASEs for the test would be something like this:

```
RVTEST_ISA("RV32IK")
...

RVTEST_CASE(0,"//check ISA:=regex(.*32.*);check ISA:=regex(.*I.*K.*);def TEST_CASE_1=True;",rol)
RVTEST_CASE(1,"//check ISA:=regex(.*32.*);check ISA:=regex(.*I.*B.*);def TEST_CASE_1=True;",rol)

...
```
Note that [RVTEST_ISA](https://github.com/riscv/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#431-required-pre-defined-macros) is currently defined to take only one argument. This argument is meant to define all the extensions (not just the base) required by the test. In RISCOF, this string is inturn added as the isa node of the [test-list.yaml](https://riscof.readthedocs.io/en/stable/testlist.html). The plugins will typically use this field as the `-march` arguments to compile individual tests.

Now based on the above example, an implementation which has implemented only "B" will enable test-case-1 but the compilation of the test may fail if the toolchain is not built with "K" since the RVTEST_ISA will drive march to be "rv32ik". 

Proposal:

The RVTEST_ISA takes multiple arguments. So for the above example, you would have the following line:

```
RVTEST_ISA("RV32IK", "RV32IB")
```

RISCOF will then check if each string is applicable under the given ISA config-yaml and appropriately build the isa string in the test-list.yaml. If only "K" is present then march=rv32ik. If both present, then march=rv32ib. This solution does impose an extra burden on the test-writer to ensure that arguments of RVTEST_ISA are in sync with the RVTEST_CASEs of the test. For eg. the test-writer shouldn't forget to add the "RV32IB" in the RVTEST_ISA if the second RVTEST_CASE exists in the above example.

This solution also doesn't require any changes to the current tests and is thus completely backward compatible and also lightweight in RISCOF. As of now, this will require only updates to the test-format spec. Once, new extensions with overlapping instructions are added, they can utilize this feature.

The above argument is also implicitly hinting that we maintain a single test for an instruction that is completely absorbed across 2 or more extensions. 

**Note**: if you are wondering why we use the RVTEST_ISA string argument to drive march instead of the ISA string in the config-yaml. Well a dut may support RV64IMAFDC. Now if we compile any of the upcoming F/D test with compressed enabled (based on the ISA string in config-yaml) the respective floating-load/stores will get compressed ( by the assembler not the compiler) and could cause changes in offsets of instructions, which may lead to signature mismatches (a false negative). Letting the RVTEST_ISA define what `-march` options to use is the safest way forward.

allenjbaum, 2021-04-27:
Are there any other cases where using the ISA string in the config-yaml may cause tests to fail?
It is possible to disable the substitution of compressed ops with " .option norvc" and enable it with " .option rvc". 
I believe (but am confirming) that this doesn't prevent using compressed ops, but prevents the assumbler from opportunistically optimizing ops to be compressed.
So, an alternative is to default all tests to run with .option norvc, which is a one line change in the .included header file.

neelgala, 2021-04-28:
We have used .option norvc in the trap-handler and in places we explicitly don't want compressed even when compressed is supported by the DUT.

Off the top of my haed, I can't seem to recollect any other corner case applicable with the current ISA. 
My only concern is : if there are new instructions (or a DUTs custom instruction) which get included in the compressed ISA tomorrow.. tests will start failing because the DUT's toolchain will compress an instruction while the SAIL's gcc toolchain may not compress that instruction. 

The reason we used the RVTEST_ISA in RISCOF is because currently each suite is only built with the respective extension (eg. [m-ext](https://github.com/riscv/riscv-arch-test/blob/b436dd0939c968f2c3da86bb9b63bb2dfe03b134/riscv-test-suite/rv32i_m/M/Makefile#L3)). Not sure what other purpose the RVTEST_ISA macro has, if not driving the march.

allenjbaum, 2021-04-30:
I have confirmed that ".option norvc" do not allow the explicit  use of any C-extension instructions  - and not just opportunistic generation.
In the future, the same may be true for other extensions that have peephole style optimization (e.g. turning a series of shifts into a B-extension instruction)
Since we can't depend on toolchains to avoid optimizations if an extension is present in the ISA string, we need to handle the restriction in the RVTESTS_ISA macro.
The intention, then, is that argument to that macro should be the architecture (RV[32 | 64 | 128][I | E] and the ISA string component for the extension being tested.
In general, there should only be a single ISA string component. But, the toolchain needs to be built with the full architecture string of the DUT, while the test must be compiled with the minimal flags needed to run the specific test.
For instructions that are shared, the tests will be officially part of one extension, and borrowed from there. If the DUT supports only the other extension, the test can't be compiled. The solution to that is to use the RVTEST_ISA macro parameter to supply a list of possible compiler flags, and the framework choose the one that is a proper subset of the architecture string - and that's what this spec change enables.
The spec should be updated to clarify this use so that future test generators conform to this. 
This is mostly a spec clarification, except we don't have a toolchain with B and K (both unratified) so when we test K-unratified, we need to build a toolchain that specifies K and supports Zbk, but the RVTESTS_ISA compiler flag will just specify K. When we get a proper toolchain that supports both, then we need to go back and add B to the ROL testss.
This will be an ongoing process; anytime an extension is added that depends on borrowed instructions from some other estension, but doesn't require that other entire extension, we'll need to add the new extension to the ISA field of tests of those borrowed instructions.
 At least, that how I think it's going to have to work.

pdonahue-ventana, 2021-04-30:
> My only concern is : if there are new instructions (or a DUTs custom instruction) which get included in the compressed ISA tomorrow.. tests will start failing because the DUT's toolchain will compress an instruction while the SAIL's gcc toolchain may not compress that instruction.

This implies that the test is assembled twice with two different toolchains so that the DUT and the reference model actually execute different ELF files.  Is that true?

allenjbaum, 2021-05-01:
If a test fails, and the test report says it wasn't compiled on the
known-good toolchain, isn't it up to the DUT to use the right toolchain?
The reasoning I've heard that  the ELFs would different is the
linker scripts could be different; but if Sail can accept any linker script
- because its memory map is generic and anything should work - then I'm not
sure why an ELF file would be different.

neelgala, 2021-05-01:
> This implies that the test is assembled twice with two different toolchains so that the DUT and the reference model actually execute different ELF files. Is that true?

Yes the ELFs for DUT and REF, in most scenarios, will be different. Some common reasons of divergence:

1. The boot-code routine for setup a hart of a DUT may not be applicable to REF
2. The sequence to terminate the test will differ between DUT and REF
3. The boot/termination-code in DUT could use some custom instructions which REF's toolchain may not be able to compile. This doesn't affect the test-section of the ELF, but the ELFs as a whole be different.

allenjbaum, 2021-05-03:
It kind of makes me want to change the way tests are laid out so that the
actual test code starts in the same location on the reference model, and
all the other stuff starts at the end of the test instead of the other way
around.
That makes a bunch of problems just go away.

pawks, 2021-09-29:
I have updated the Specification based on this proposal over [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/riscof-dev/spec/TestFormatSpec.adoc#431-required-pre-defined-macros). To find an example of how the new macro works with the framework check [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/riscof-dev/spec/TestFormatSpec.adoc#appendix-b-example-illustrating-the-use-of-multiple-isas-for-a-test).

Riscof has also been updated to work with this on [this branch](https://github.com/riscv-software-src/riscof/tree/arch-test-release/riscof).

allenjbaum, 2022-12-08:
This is now complete

closed 2022-12-08 by allenjbaum