riscv/riscv-arch-test #190: the 16-byte signature boundary issue

neelgala, 2021-04-27:
Consider the [div-01.s](https://github.com/riscv/riscv-arch-test/blob/master/riscv-test-suite/rv32i_m/M/src/div-01.S) test which has 590 (starts with inst_0) test-cases. So the signature region created is 590*4 bytes = 2360 bytes in size initialized to 0xdeadbeef. **Now this is not a multiple of 16**.

Now if we go back to the test-format-spec here it says: 
- `the signature should always begin at a 16-byte` 
- `"it should also end at a 16-byte boundary`. 
Due to the above requirements,  the [spike model_test.h](https://github.com/riscv/riscv-isa-sim/blob/9d4f45c2ebf105503974fc80a42590ca1584c354/arch_test_target/spike/model_test.h#L32) constraints the alignment of the the RVMODEL_DATA_BEGIN and RVMODEL_DATA_END to 16-byte boundaries. This means the RVMODEL_DATA_END is at an offset of 2368 bytes from RVMODEL_DATA_BEGIN. 

Therefore, there are 8 bytes of data in the signature which are not initialized by the test and can be any value the model decides (which is you use GCC is always 0x0). Hence, if you look at the last [2 entries of the reference signature](https://github.com/riscv/riscv-arch-test/blob/b436dd0939c968f2c3da86bb9b63bb2dfe03b134/riscv-test-suite/rv32i_m/M/references/div-01.reference_output#L591-L592) of div-01.reference_output they are going to 0's instead of 0xdeadbeef

Now, since the extra padded values are controlled by the DUT, they can be anything. 

Possible solutions:

1. the spec needs to be updated to mandate that the values being padded to meet alignment must be 0xdeadbeef. It shouldn't be 0x0 since the spec mentions [here ](https://github.com/riscv/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#451-common-test-format-rules)that the signature must be loaded with 0xdeadbeef This will require updating the references and all ported targets. The targets could achieve the above by simply using `.align 4,0xdeadbeef` instead. No changes to tests.

2. Change the spec to mandate that the signature should end at a 4-byte boundary. This will require all targets to remove any `.align 4` from their macros, since by default all test signatures end at a 4-byte boundary. This will require references to be updated. No change to tests.

3. Explicitly padd the tests to align at 16-byte boundaries. This will require updating the tests and the corresponding references. No changes to the model side (or atleast that is what I hope).

Would like to know which of the above is best suited? or if there is another solution.

allenjbaum, 2021-04-27:
To be clear: this is a very subtle test format spec bug: the 16byte alignment (an artifact that is no longer needed, really) colliding with the the user defined data area that has no boundary restriction, and no initialization requirement for either the end of the tset or the beginning of the user defined area.

neelgala, 2021-09-10:
Considering this is solved after we transition to RISCOF - option 1 and 2 seem the least burdensome - since a lot of folks will be re-writing their targets and RISCOF will no longer require hosting references. So its only the targets making changes. 

Any thoughts on what would be the best way forward ?

allenjbaum, 2021-09-12:
Both #1 and #2 require changes to the spec.
#1 requires changes to all targets as well. Doesn't #2 also require changes
to targets?

neelgala, 2021-09-12:
yes the option-2 required the targets to remove `.align` from the model_test.h files.

allenjbaum, 2021-09-21:
Besides Seagate - who has complained that something is broken?
If no one has, (and Seagate has managed to either fix or work around the
issue) and everyone must change their target to riscof plugins andyway,
then let's both
 - change the spec to remove the *requirement*  for padding AND
 - require that, if there is padding, that it be initialized to deadbeef.
How does that sound? We could probably write a script that modifies every
existing test to remove the .align and/or add padding initialization, and
update all the existing tests with a single PULL

We need to make a list of changes that need to be made to the test format
spec going forward.
This is one; others are:
1. the addition of a new label at the start of the trap signature area
(used in the trap handler boot code at least, but must be used inside
RVMODEL_DATA_BEGIN/END)
2. the addition of variables that define whether trap handlers need to be
instantiated for S and VS modes (used in the trap handler)
3. new rvmodel macros for setting and clearing interrupts for various modes
(need to coordinate with AIA and CLIC specs)
4. new rvmodel macros for asynch events: interrupts and memory read/write
(we still need to discuss this, or form a TG to define it)

allenjbaum, 2022-02-09:
1.The updated trap handler has the label mtrap_sigptr to indicate the start of the trap signature area.
We may also need Strap_sigptr and Vtrap_sigptr to handle virtual memory - TBD

2. Those variables are called rvtest_strap_routine and rvtest_vtrap_routine

allenjbaum, 2022-08-25:
Close after updating and merging tests that substitute rvmpodel_data_begin_end with rvtest_sig_begin/end

allenjbaum, 2022-12-08:
partially closed with PR#286 "Cleanup", now needs documentation update

commit 5cbc33a3 (2022-12-17): Fix #190. Drop 16 byte alignment requirements for signature region and change size to multiple of 4.
commit 3443cb28 (2022-12-17): Fix #190. Modify signature size requirement to multiple of 4.
commit 146abcb9 (2022-12-17): Fix #190. Modify signature size requirement to multiple of 4.
PR #302 (2022-12-27): Canary patch
## Description
- Fix Canary definition to depend on sigalign.
- Fix SIGALIGN definition to work in all cases of FLEN and XLEN definitions.
- Fix inconsistencies in tests (missing trap regions and incomplete ISAs).
- Add Zicsr to ISA in RV32 privilege tests
- Modify signature size requirement to multiple of 4.

### Related Issues
- Fixes #190 
- Fixes #280
- Fixes #299
closed 2022-12-27 by neelgala