riscv/riscv-arch-test #191: Asking for help porting to a new CPU design

jeras, 2021-05-02:
I am writing my own hoby RISC-V CPU and I would like to start running compliance tests.
Right now I am trying to compile the tests with my linker script [`link.ld`](https://github.com/jeras/rp32/blob/4b5c3e03fbafe5643e5e8df6d37e75c28328d734/test/r5p/link.ld) script and header file [`model_test.h`](https://github.com/jeras/rp32/blob/4b5c3e03fbafe5643e5e8df6d37e75c28328d734/test/r5p/model_test.h).
I know well enough what a linker is supposed to do, but I do not have experience with linker file syntax.
The other source of my confusion are examples, which are mostly migrations from v0.1 to v0.2, I tried to follow the v0.2 approach.

I wrote a split instruction (`mem_if` ()instruction fetch) and data (`mem_ls` (load/store)) memory linker file.
Code should go into `mem_if`, the rest into `mem_ls`, including initialized variables and data constants, since I plan to initialize both memories in the Verilog testbench.

In the header file, I kind of tried the HALT code, but with a different source file compiled with LLVM. My point is, this part of the code should make sense, but I am not sure about the other macros.

Compliance test repo is a submodule, I created the target folder structure in my repo, and did not modify any files in the compliance source. I call make with a path and provide  a set of environment variables.

To reproduce this issue do:
```bash
git clone https://github.com/jeras/rp32.git --recursive
cd rp32
git checkout 4b5c3e03fbafe5643e5e8df6d37e75c28328d734
cd test
./test.sh
```

The error I get is:
```
... riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 0000000000003280
```

neelgala, 2021-05-03:
you will need to change _start in your[ link.ld ](https://github.com/jeras/rp32/blob/4b5c3e03fbafe5643e5e8df6d37e75c28328d734/test/r5p/link.ld#L3) to rvtest_entry_point.

That should solve the error you are currently facing.

jeras, 2021-05-03:
Thanks @neelgala I made a bit of progress with the sugested change.
I found a bug in the example Makefile.include, there is a missing semicollon:
https://github.com/riscv/riscv-arch-test/blob/master/riscv-target/example-target/device/rv32i_m/Makefile.include#L23
https://github.com/riscv/riscv-arch-test/blob/master/riscv-target/example-target/device/rv64i_m/Makefile.include#L23

A few tests compile properly, now I get the next issue:
```
...lowrisc-toolchain-gcc-rv32imc-20210412-1/bin/../lib/gcc/riscv32-unknown-elf/9.2.0/../../../../riscv32-unknown-elf/bin/ld: address 0x37218 of /home/ijeras/VLSI/rp32/submodules/riscv-arch-test/work/rv32i_m/I/beq-01.elf section `.text.init' is not within region `mem_if'
```

stnolting, 2021-05-03:
> A few tests compile properly, now I get the next issue:

Maybe you have forgotten to map `.text.init` to the according (instruction?) memory in your linker file.

I have also [ported](https://github.com/stnolting/neorv32/tree/master/riscv-arch-test/port-neorv32/framework_v2.0/riscv-target/neorv32) the architecture tests for a custom RISC-V CPU. For the `I` tests I am using a split memory system: One large memory array (starting at `0x00000000`) for instructions and another small one (starting at `0x80000000`) for data. My according linker script looks like this:

```
OUTPUT_ARCH( "riscv" )
ENTRY(rvtest_entry_point)

SECTIONS
{
  . = 0x00000000;
  .text :
  {
    *(.text.init)
    . = ALIGN(0x1000);
    *(.text)
    . = ALIGN(0x1000);
    _end = .;
  }

  . = 0x80000000;
  .data :
  {
    . = ALIGN(0x1000);
    *(.tohost)
    *(.data)
    *(.data.string)
    *(.bss)
  }
}
```

jeras, 2021-05-03:
Thanks @stnolting, I used your linker file and it works.
I will probably use it, at least for now.
But I would also like to learn how to write link files, and here the one based on the example:
https://github.com/riscv/riscv-arch-test/blob/master/riscv-target/example-target/env/linksplit.ld
This one properly specifies memory sizes, so the linker can report errors, if there is not enough ROM for example.

Linker files kind of remind me of [devicetree](https://www.devicetree.org/) files, you spend a year trying to understand the syntax well enough to write sensible files, then it takes a month to forget.

allenjbaum, 2021-05-06:
Can this be closed if you have your answers?

jeras, 2021-05-07:
Yes this can be closed, I was able to run the tests, and my CPU is passing RV32I tests now.

Still the provided example is buggy, and can't be executed (make compile) out of the box.
I know an entirely new architecture test framework is being developed, so I guess it does not make sense to spend too much time on this one.

closed 2021-05-07 by jeras

neelgala, 2021-05-07:
@jeras thanks for the catch. I'll try to patch this up in the next PR.

MarcKarasek, 2021-05-07:
The example was never meant to compile.  It is a template to create your
own  target.

Sent from Nexus 7
Marc Karasek
92-57 Bear
Troop 648 Scoutmaster