riscv/riscv-arch-test #193: CHECK_XLEN macro will silently PASS

odarcy1, 2021-06-18:
I ran the test rv32ui-p-addi on a 64bit machine using spike --isa=rv64ima

I noticed that the CHECK_XLEN macro defined in env/p/riscv_test.h will yield a PASS result. I am not sure that is the correct behavior ?
If the goal of this macro is to check that the HW register length matches the isa/abi with which the test was written/compiled , then shouldn't it report a failure instead ?

# define CHECK_XLEN li a0, 1; slli a0, a0, 31; bltz a0, 1f; RVTEST_PASS; 1:

On a 64bit core, the result of the slli instr will be 0x00000000_80000000
then the bltz instr will not take the branch because a0 is a positive number, therefore it will execute the RVTEST_PASS macro and exit prematurely while skipping the "addi" portion of the test.

Indeed the same result occurs for any rv32ui-p-* tests. It will prematurely exit with a PASS status.

I think it should exit right away with a FAIL status, so that the user knows something is wrong as he is trying to run some 32bit tests on a 64bit machine.

If someone can shed some light on this please...it we would be greatly appreciated.

neelgala, 2021-06-19:
@odarcy1 I believe you are working on tests which are hosted here : https://github.com/riscv/riscv-tests and not the ones hosted in this repository. If so, it would be best to move this issue to the riscv-tests repository.

allenjbaum, 2021-07-01:
I'm going to resolve this unless I hear objections in the next 30 days. This is not a spec'ed  riscv-arch-test macro and not found in this repo.

closed 2021-08-17 by allenjbaum