riscv/riscv-arch-test #200: Invalid riscv-test-suite/rv32i_m/privilege/references/ebreak.reference_output

ppaulweber, 2021-08-03:
It looks like the provided reference signature for the `ebreak` instruction in the `rv32i_m` test suite was generated either with an outdated `spike` [1] or `sail-riscv` [2] simulation, because I've just detected the same invalid `ebreak` behavior in a hand-written RISC-V simulator that the `mtval` register is set incorrectly to `PC` whereas it shall be set to zero, because the `ebreak` instruction is not part of setting the `mtval` to PC and belongs to the "other traps" as specified in the RISC-V specification [3, p. 38]:

> When a hardware breakpoint is triggered, or an instruction-fetch, load, or store address-misaligned, access, or page-fault exception occurs, mtval is written with the faulting virtual address. On an illegal instruction trap, mtval may be written with the first XLEN or ILEN bits of the faulting instruction as described below. For other traps, mtval is set to zero, but a future standard may redefine mtval’s setting for other traps.

I've execute the `ebreak` test [4] twice in my hand-written simulator:
(1) setting `mtval = PC` during the execution of the `ebreak` instruction (see left-hand-side of the following signature output); and 
(2) setting `mtval = 0` during the execution of the `ebreak` instruction (see right-hand-side of the following signature output):

~~~
// mtval = pc   | // mtval = 0
00000000        | 00000000
11111111        | 11111111
0000008f        | 0000008f
00000003        | 00000003
00000108        | 00000108
00000108        | 7ffffff4
00000000        | 00000000
00000000        | 00000000
~~~

The current upstream `ebreak` signature reference output matches the invalid `mtval = PC` behavior:
https://github.com/riscv/riscv-arch-test/blob/20f58fc63eaa60d56036690ca40d9008a3942ea8/riscv-test-suite/rv32i_m/privilege/references/ebreak.reference_output#L1-L8

To be compliant with the RISC-V specification [3], the signature reference output at Line 6 should be changed to `7ffffff4`:
https://github.com/riscv/riscv-arch-test/blob/20f58fc63eaa60d56036690ca40d9008a3942ea8/riscv-test-suite/rv32i_m/privilege/references/ebreak.reference_output#L6-L6

[1] https://github.com/rems-project/sail-riscv/issues/44
[2] https://github.com/riscv/riscv-isa-sim/issues/426
[3] https://github.com/riscv/riscv-isa-manual/releases/download/Ratified-IMFDQC-and-Priv-v1.11/riscv-privileged-20190608.pdf
[4] https://github.com/riscv/riscv-arch-test/blob/master/riscv-test-suite/rv32i_m/privilege/src/ebreak.S

neelgala, 2021-08-03:
Sail was also subsequently changed to match this behavior.

ppaulweber, 2021-08-03:
Dear @neelgala, 
thank you for the quick reply and providing me with the follow-up discussions. In this case I can finalize the `mtval = PC` behavior for the `ebreak` instruction in my implementation as well.

Maybe, a short hint on the README page that this RISC-V tests follow a specific RISC-V ISA manual version would help to avoid future issues, because I assumed that the provided tests in this repository follow the latest available specification on the RISC-V website [1] which is still the "Volume 2, Privileged Spec v. 20190608" [2].

[1] https://riscv.org/technical/specifications
[2] https://github.com/riscv/riscv-isa-manual/releases/download/Ratified-IMFDQC-and-Priv-v1.11/riscv-privileged-20190608.pdf

neelgala, 2021-08-03:
@ppaulweber yes.. exposing the riscv-arch-test sync with specific isa and toolchain version is on the TODO list of the SIG. We should get around to it soon. Closing issue for now.

closed 2021-08-03 by neelgala