riscv/riscv-arch-test #202: Riscv compliance test (I-FENCE.I-01.S) corrupting the .text region

dineshannayya, 2021-08-20:
I see I-FENCE.I-01.S  fence.i instruction corruption/updating content in .text region.  
.text.init and .text are in ROM area, How to move this logic to RAM area (.data segment) ?

 364:	0000100f          	fence.i

00000368 <instr_A_dst>:
 368:	00000137          	lui	sp,0x0
 36c:	0018a023          	sw	ra,0(a7)
 370:	0028a223          	sw	sp,4(a7)
 374:	0038a423          	sw	gp,8(a7)
 378:	00f8a623          	sw	a5,12(a7)

neelgala, 2021-08-20:
I would first suggest picking the latest set of tests which is named [Fencei.S](https://github.com/riscv/riscv-arch-test/blob/master/riscv-test-suite/rv32i_m/Zifencei/src/Fencei.S). 

The test basically checks if the updates to the instruction memory are properly synchronized with the data memory (which is the primary objective of the instruction Itself). Hence, the test does perform store in the .text regions. 

If you have implemented fence.i in your DUT, then to pass this test you will need to load the program in a read-write section of the your test-bench memory. This is specific to your DUT and testbench environment. The test itself doesn't require any changes.

allenjbaum, 2021-09-06:
Im going to close this as it appears not to be a test issue; if you disagree, reopen it

closed 2021-09-06 by allenjbaum