riscv/riscv-arch-test #203: Fence test has poor coverage

allenjbaum, 2021-08-24:
Fence has an optional .TSO extension.
The fm field of the instruction is normally 0000, but is 1000 for the .TSO extension (specifically with P=0011 and S=0011).
The instruction description says that other encodings are reserved -- but then says that reserved values should be ignored,  which is different than normal reserved fields which tests avoid because we can't count on what happens.

So, the fence.01.S test needs to expanded to at least have a walking ones test for the fm field.

aswaterman, 2021-08-28:
There's no reason to wait for an assembler change.  Just use the `.word` directive and manually assemble the instruction.

In fact, I'm not sure the assembler change will ever come; the assembler maintainers might (legitimately) balk at adding support for assembling invalid instructions.  `.word` will always work.

allenjbaum, 2021-08-29:
Let's go with Andrew's suggestion then.
He also pointed out that for SFENCE.VMA and friends, the MSBs of rs2 are reserved for future use, should be zeroed by software, and should be ignored by current implementations. We don't have tests for that yet, but we need to note it somewhere for when we do.

allenjbaum, 2022-08-25:
Resolve to write more tests; bring up to dev partners, and close when tests are merged.

allenjbaum, 2023-02-09:
There is a lot of overlap here with issue #119

allenjbaum, 2023-02-11:
The syntax for a fence instruction with arbitrary params is:
.insn i 0x0F, 0, rd, rs1, (signed_fm<<8) | (pred<<4) | succ )
where legal values of fm are 0, -8, 

The syntax for a  fence.i instruction with arbitrary params is:
.insn i 0x0F, 1, rd, rs1, signed_imm12

The syntax for an sfence.vma instruction with arbitrary params is:
.insn r 0x73, 0, 0x09, 0, rs1, rs2