riscv/riscv-arch-test #206: Reference word for test 'rv32e_unratified/C/cswsp-01' is address-dependent

cgroux, 2021-09-10:
I've run all 'rv32e_unratified' tests, from folders 'C' and 'E'.
All reference values, except the one I mention, are independent of the absolute mapping of the memory.

However, in case of the 'cswsp-01' test, at label 'inst_3:', the code stores its current stack pointer (x2) into the signature area:

Here is the source code, from [cswsp-01.S](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/riscv-test-suite/rv32e_unratified/C/src/cswsp-01.S):

```inst_3:
// rs2==x2, rs2_val == -536870913, 
// opcode:c.swsp; op1:x2; op2:x2; op2val:-0x20000001; immval:0x2c
TEST_STORE(x4,x5,0,x2,x2,-0x20000001,0x2c,12,c.swsp,0)
```
And the corresponding disassembly, for my platform:

```000000ca <inst_3>:
  ca:	e0000137                lui	sp,0xe0000
  ce:	fff10113                addi	sp,sp,-1 # dfffffff <fromhost+0xddfffaff>
  d2:	00c20113          	addi	sp,tp,12 # c <rvtest_entry_point+0xc>
  d6:	02c00293          	li	t0,44
  da:	40510133          	sub	sp,sp,t0
  de:	d60a                	sw	sp,44(sp)       <-- Saves a memory-mapping-dependent value into the signature
  e0:	0001                	nop
  e2:	0001                	nop
```
As consequence, the signature for this particular test will only match the reference (0x800021e4) for a precise memory layout. 

While this is not strictly-speaking a bug, it would be cleaner if all tests are performed in a way that do not depend on absolute memory adresses.

pawks, 2021-09-10:
This seems to be an error in the coverage definition for the `c.swsp` instruction. `rs2==x2` should not be considered as a coverpoint for `c.swsp` due to it being used as the base address i.e `rs1` in this case.

cgroux, 2021-09-10:
Thanks for the reply. I am not familiar with 'coverpoint', and how it is defined, or when it is applied. For now, I am just comparing raw signature from RAM with '.reference_output' files. Maybe there is some scripts above, which I don't know and don't call, which would discard the mismatch I observe. If so, then I apologize for this irrelevant issue report.

pawks, 2021-09-10:
The issue is completely relevant. The test instance generated is wrong and hence the source files used for generating these tests will have to be changed. The tests and the coverage reports will have to be regenerated to fix this issue.

neelgala, 2021-09-10:
@cgroux thanks for reporting this. What pawan was referring to is that we use coverpoints to generate tests using CTG. So effectively the coverpoints used to generate this particular tests were wrong.

@bilalsakhawat-10xe This [line](https://github.com/riscv-software-src/riscv-ctg/blob/master/sample_cgfs/rv32ec.cgf#L454) needs to be `rv32e_regs_mx2` as is done [here](https://github.com/riscv-software-src/riscv-ctg/blob/6838350110ae026fc20d9490d6ec287aa3ff7763/sample_cgfs/rv32ic.cgf#L454).  

I would suggest fixing the cgf in ctg, and then regenerating this test and all its collaterals as a separate PR.

bilalsakhawat-10xe, 2021-09-10:
@neelgala sure will do ASAP.

neelgala, 2021-10-01:
fixed in #210