riscv/riscv-arch-test #208: typo in example target code

jeras, 2021-09-12:
Should be `.align` instead of `.alive`.
https://github.com/riscv-non-isa/riscv-arch-test/blob/9141cf9274b610d059199e8aa2e21f54a0bc6a6e/riscv-target/example-target/model_test.h#L27

Somehow I am using this code with the typo but not having any issues.

neelgala, 2021-09-13:
The description of `.align` is as per the riscv-asm/gnu pseudo ops defined [here](https://github.com/riscv-non-isa/riscv-asm-manual/blob/master/riscv-asm.md#pseudo-ops) 

It is meant to align the end of the signature at a 16byte boundary has demanded in the [Testformat spec](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#36-the-test-signature)

jeras, 2021-09-13:
It is not `.align` there is a typo, it is written **alive**, as in all the living creatures, please look at line 27 again.
Since this example is not compiled and executed, there are probably more such typos.

neelgala, 2021-09-13:
my bad.. should I have read your comment better.. will fix it.

commit 4f6f3c58 (2021-09-13): fixed typo `.alive` --> `.align` in `riscv-target/example_target/model_test.h`.

close #208
closed 2021-10-04 by neelgala