riscv/riscv-arch-test #209: Difficulty compiling 32-bit tests

davidharrishmc, 2021-09-13:
I'm having trouble finding documentation on how to compile 32-bit versions of the tests.  Sorry this seems like a dumb question, but it likely affects other new users too.

Per the instructions, I've changed export XLEN to 32 in Makefile.include.

running make fails to find riscv32-unknown-elf-gcc.  I have compiled gcc with default settings, which just produces riscv64-unknown-elf-gcc.  This week, Neel verbally recommended changing riscv32-unknown-elf-gcc to riscv64-unknown-elf-gcc in riscv-target/spike_0p1/device/rv32i_m/I/Makefile (and sounded like that might be something he would change in the repo to play nicely with default gcc installations).  However, even with that change, make still seems to be looking for riscv32-unknown-elf-gcc and I'm not sure where it's finding that.

Compile /home/harris/github/riscv-arch-test/work/rv32i_m/C/cadd-01.elf
/bin/sh: riscv32-unknown-elf-gcc: command not found
\e[31mriscv32-unknown-elf-gcc failed for target /home/harris/github/riscv-arch-test/work/rv32i_m/C/cadd-01.elf \e[39m

neelgala, 2021-09-13:
@davidharrishmc Which target are you using ? If you are using SAIL then [this](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/riscv-target/sail-riscv-c/device/rv32i_m/C/Makefile.include#L12) line needs to change to `riscv64-unknown-elf-` and similar changes in all other files.

For spike similar change needs to be made [here](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/arch_test_target/spike/device/rv32i_m/C/Makefile.include#L7).

Are you still facing the error after making these changes ?

pawks, 2021-09-13:
Alternatively you can add the following line to the [Makefile.include](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/Makefile.include) file present in the root directory:
```
export RISCV_PREFIX = riscv64-unknown-elf-
```

MarcKarasek, 2021-09-13:
You need to compile the 32 bit version of gcc.  In order to do this :

# Build the 32 bit toolchain
./configure --prefix=/opt/riscv32 --with-arch=rv32gc --with-abi=ilp32d
--with-cmodel=medany
sudo make newlib -j 4
sudo make linux -j 4

run these commands in the riscv-gnu-toolchain dir you cloned from the riscv
github repo.

I am assuming from your post that you have already run the compile for the
64 bit version of the tools.  But just in case here is the commands for
that:
./configure --prefix=/opt/riscv64 --with-cmodel=medany
--with-arch=rv64imafdc --with-abi=lp64d
sudo make newlib -j 4
sudo make linux -j 4

I have a couple of scripts to add the toolchain to my PATH
export PATH="$PATH:/opt/riscv64/bin"
export RISCV="/opt/riscv64"

export PATH="$PATH:/opt/riscv32/bin"
export RISCV="/opt/riscv32"

You can put these in a script and run them each time you compile or put
them into your .bashrc file.

I also infer from your post you are building the old riscv-arch-test, I
would encourage you to drop this and instead focus your work on the new
riscof framework.

You can find the docs on how to install and setup riscof here:
https://riscof.readthedocs.io/en/stable/intro.html
We should have an example project up on github soon that can be used as a
template for riscof..

Marc

After this is done you will have two dir under /opt on your system.  One
for the 32 bit version of the compiler, one for the 64 bit version.

davidharrishmc, 2021-09-13:
Neel,

I was running default out of the box, so spike.  Your fix took care of it.  I wouldn’t have guessed to look in riscv-isa-sim to change the compile behavior of riscv-arch-test.

Thank you,

David

MarcKarasek, 2021-09-13:
Neel told me you are on CentOS.  Which version?  And what are your future plans?  Rocky Linux??

davidharrishmc, 2021-09-13:
We’re on CentOS 7.6 because the server also runs the Cadence and Synopsys suite, which list RedHat for support.

We will stick on this as long as we can (hopefully several years), and then follow the guidance of Cadence and Synopsys if their tools cease to support CentOS 7.  Rocky Linux may be the next step.

David

davidharrishmc, 2021-09-13:
> I was afraid of that..
> CentOS 7.x is based on RHEL 7 and the newer framework is very difficult to near impossible to get installed on. This is due to a number of issues, supported compiler tools, etc. under RHEL7/CentOS7.
>
> The older framework can be used for in-house validation, but I believe there is no plan currently to allow it to be used for getting RISC-V branded. For that you would need to use the newer riscof (python) based framework.
>
> Marc

Is the latest Ubuntu our best bet?

MarcKarasek, 2021-09-13:
Ubuntu is not supported by any of the tools vendors cadence, mentor, etc...

You're best bet would be rhel8 or rocky Linux.  Centos8 is only be
supported through the end of this year.  So it may not be a valid choice.

You should join us for a meeting if the arch test group.

Is your company a risc-v member?

neelgala, 2021-09-19:
I am closing this as the issue is resolved.

closed 2021-09-19 by neelgala