riscv/riscv-arch-test #211: default rvtest_data should be 16-bytes

neelgala, 2021-09-19:
Majority of the tests include a default `rvtest_data` label which contains the single word `0xbabecafe`. This was meant to be used by the load-store tests. However, `ld` tests load the upper bits as 0x0 - which does break anything but is definitely now the desired behavior of the test. 

Fix would be ensure that the default rvtest_data section has 16 bytes of data (rotated versions of 0xbabecafe). This will also ensure that any other data following it will be starting from at a 16-byte alignment as well

neelgala, 2022-05-26:
This particular suggestion will require all tests to change. is that acceptable ?

allenjbaum, 2022-05-27:
An alternative is to make sure the data section ends at a 16 byte alignment
and have the boot code manually store 0xdeadbeef or whatever from that
label until
the previous 16B alignment ( or RVMODEL_DATA_END, whichever comes. That
would be compatible with both old and new, wouldn’t it? The only file that
would be the .h file.

closed 2022-11-04 by pawks