riscv/riscv-arch-test #212: RISCV-Target don't have swerv target

YeuzhiHere, 2021-09-21:
Swervolf EH1 is very popular RISC-V based core. But In this repo, there is no target for swervolf EH1. Is there any specific reason? 
Kindly resolve this issue.

bilalsakhawat-10xe, 2021-09-21:
The popularity of a RISC-V based core is irrelevant for it to be a RISC-V Target. If sweRV EH1 is not a RISC-V target its not an issue. But if you want to validate sweRV EH1 core or use it as a RISC-V Target, you can look into [Connect Riscof to SweRV EH2 core](https://github.com/riscv-software-src/riscof/issues/10) issue.

allenjbaum, 2021-09-21:
Targets are the responsibility of the model owners, not RISC-V.
If you're developing a SWERV based core, you can create a target directory
just as anyone else would do for their own implementation, and run the
existing architectural compatibility tests on it.
When V3 of the test framework becomes available, all target directories
will be moved into the repos of the models themselves.
This includes the 2 RISC-V sanctioned reference models, SAIL and SPIKE.
Bilal's link above is a good place to start.

YeuzhiHere, 2021-09-21:
Thanks both of you :)

closed 2021-09-21 by YeuzhiHere