riscv/riscv-arch-test #213: Running Compliance Test for Swerv

YeuzhiHere, 2021-09-24:
**Note: I posted this issue in chipalliance/Cores-Swervolf repo as well. Posting here, because its quite active repo. Hope everyone will understand.**

I am trying to do compliance test for swerv core using the mentioned guideline, but my result is;

[image]

I am new to compliance thing and I will post my each step so that understanding of the whole scenario is clear.

**1.** I am following this section below;
[image]

**2.** When I acted exactly as mentioned in this section, I got these directories (exception is for gnu-riscv-toolchain, which I added myself)
[image]

**3.** This is my riscv-target directory;
[image]

**4.** Upon running this thing as it is, I got few errors like there is no directory named rv32i_m/rv32i in riscv-test-suite. So I changed makefile.include, makefile in riscv-compliance directory and also added rv32i directory in riscv-test-suite. Below are pictures;
[image]
[image]
[image]
 
**5.** I also edited makefile.include in riscv-target/swerv/device/ directory, which is as under;
[image]

**6.** I also got error that model_test.h file not found, so I copied it from some example folder of **riscv-arch-test repo** (might be I did wrong here). Here is pic;
[image]

**7.** Finally here is my work/rv32i_m/rv32i directory;
[image]

**8.** Kindly do guide me and resolve this issue. You can suggest me any other simulator or any repo (where you have done swerv compliance test). If i am missing any detail, plz do mention.

YeuzhiHere, 2021-09-24:
Another thing, nothing is being written to my **signtaure.output** file

neelgala, 2021-09-24:
looks like this documentation that you are following is for an older vrsion of the test-framework. Remember you should not be changing anything in the riscv-test-suite directory at all. All changes are required to happen in your riscv-target/swerv/ directory only. You can follow the guide mentioned here : https://github.com/riscv-non-isa/riscv-arch-test/tree/master/doc#porting-a-new-target to port your swerv target for the current framework.

YeuzhiHere, 2021-09-27:
Thanks @neelgala for your valuable comment. I will follow your recommendations.

closed 2021-09-27 by YeuzhiHere