riscv/riscv-arch-test #220: F tests

davidharrishmc, 2021-10-20:
riscv-arch-test/riscv-test-suite/rv32i_m now has the F directory of tests.

When doing a Make with RISCV_TARGET ? = spike

the riscv-arch-test Makefile looks in riscv-isa-sim/arch_test_target/spike/device/rv32i_m for the RISCV_ISA_ALL.  There is no F directory in arch_test_target, so the F tests don't get made.

I could manually modify arch_test_target to add a F directory, but is there a more graceful solution such that the F tests get made "out of the box"?  Does riscv-isa-sim need an update?

Thank you,

David Harris

davidharrishmc, 2021-10-20:
> @davidharrishmc <https://github.com/davidharrishmc> yes.. the arch_test_target directory in riscv-isa-sim needs an update. It was on my todo list, but haven't gotten around to raising that PR.
>
> In the current framework the solution is to add an "F" directory under device/rv32i_m/ - with obvious changes in the spike --isa arguments in the Makefile.include file.

Thanks!

allenjbaum, 2021-11-23:
Is this ready to be closed, or should it wait until thearch_test_target directory in riscv-isa-sim is updated?

davidharrishmc, 2021-12-01:
Allen,

Not sure if this was a query for me.

I’d love to see riscv-isa-sim arch_test_target updated.

In the meantime, my scripts are manually patching the rv32i_m/F and r64i_m/D makefiles.  Perhaps we will need the same soon for rv64i_m/D and rv32i_m/F.

I don’t know the politics on riscv-isa-sim but am happy to submit pull requests for those Makefiles if it is helpful.

Thanks!

David

allenjbaum, 2021-12-15:
This will get fixed shortly, and closed, I've been told...

allenjbaum, 2022-03-30:
Do we want to fix this, or just wait for the cutover to Riscof?

davidharrishmc, 2022-03-30:
Closing this; will look at riscof.

closed 2022-03-30 by davidharrishmc