riscv/riscv-arch-test #222: Exceptions mtval register value (zero or informative)

fborisovskii, 2021-11-18:
Hello,

We've found some tests shows our implementation incompatible when mtval is set to zero after exception.

Current privilege specification allows hardware platform to determine which exceptions set mtval informatively
and which set it to zero. So I suggest it is better to check if exceptions mtval value is zero or informative 
and accept both cases as compliant.

Priv spec (https://github.com/riscv/riscv-isa-manual/blob/master/src/machine.tex):
The hardware platform will specify which exceptions must set {\tt mtval} informatively and
which may unconditionally set it to zero.

For example our HW platform sets mtval to zero when breakpoint exception occurs (I don't see necessity to duplicate mepc to mtval in that case).

Best regards,
Fedor

marcfedorow, 2021-11-18:
TLDR: changes described in #200 should be reverted while [this](https://github.com/riscv/riscv-isa-manual/commit/51522f98b5cae989c7d161e00b20a08483a2301e) is a backwards compatible normative change, since writing 0 to mtval remains a legal option.

allenjbaum, 2021-11-18:
This is one of the cases that the existing framework cannot handle; it has
a single reference signature, and that just doesn't work if there are two
(or more) legal results (which virtually any WARL enables).

The riscof arch-test framework was created mostly to solve that problem. It
will configure the SAIL reference model to match the legal architectural
options that a device implements and create a reference signature that
matches the capabilities and options supported by the device.
It does this by reading a device specific YAML file which describes all
architectural options, and one of those options is which xcause values will
write xtval.
Documentation on riscof (which hasn't been deployed yet, but is functional)
can be found here:
      https://github.com/riscv-software-src/riscof/tree/master/docs

Note that this will require you to create the YAML file, and details of
that can be found here:
  https://github.com/riscv-software-src/riscv-isac/tree/master/docs

Note that Sail does not currently have the ability to configure itself from
the YAML; that support is coming soon.

marcfedorow, 2021-11-18:
Anything could be solved with arithmetic.
If test can not handle a compliant architecture, it obviously should not be used for compliance testing.

Please remove this check and any other check that is mentioned in mtval-related issues.

fborisovskii, 2021-11-18:
> This is one of the cases that the existing framework cannot handle; it has a single reference signature, and that just doesn't work if there are two (or more) legal results (which virtually any WARL [...]

Thank you. I will check that proposal. And I agree about WARL. But in case of exceptions mtval. Why it is bad solution to just save expected value in known place in data section and in case mtval is zero - take this value instead mtval? Thus zero mtval and expected mtval will both give the same result. In case of ebreak it is possible to take mepc instead mtval if mtval is zero.

But I agree it is better to check particular HW to be compliant to what is claimed.

allenjbaum, 2021-11-19:
You're asking for a special version for tests that may store zero or store
a real value, and the existing framework makes no distinction or way of
labelling those signature values.
Again - it is a recognized problem and is being fixed, just not quickly
enough. We could modify the framework and tests to do as you say, but we
made the decision to spend our resources on a more general fix - and that
is the riscof framework.

There was a deliberate choice to select one of the two possible results in
the current framework. The alternative - to not test MTVAL at all - was
deemed unacceptable.
So, no, that check will not be removed.
It will be fixed in the future, as explained in the previous email.

marcfedorow, 2021-11-19:
> So, no, that check will not be removed.

I am definitely **not** saying "remove this check forever".
Still this check does not work for now.
Obviously it is not that hard to remove `rv32i_m/privilege/Makefile.include` untill mtval issues
> will be fixed in the future, as explained in the previous email

, so it does not require that much of resources to make any decisions on how to spend them.

If tests does not work on a compliant architecture, they should not be included in a **default** run.

allenjbaum, 2021-11-19:
If we do that, we fail to test other architectural compliant
implementations, so no.
Run your tests, you'll see a failure, you know what it, and if you intend
to use that as a part of the self-certification, you can ask for a waiver
(see the Architectural Test policy) and you'll get it.

fborisovskii, 2021-11-23:
I am closing this issue. Allen, thank you for reply. riscof  looks promising. I think it will become an answer to this issue. 

Best wishes,
Fedor

closed 2021-11-23 by fborisovskii