riscv/riscv-arch-test #223: fcsr reset value

fborisovskii, 2021-11-23:
Hello,

I've watched objdump of F tests and found that fflags part of fcsr is not set before the first test run.
```
80000218 <rvtest_code_begin>:                                                                                      
80000218:       00006137                lui     sp,0x6
8000021c:       300021f3                csrr    gp,mstatus
80000220:       0021e133                or      sp,gp,sp                                                           
80000224:       30011073                csrw    mstatus,sp                                                         
80000228:       00004817                auipc   a6,0x4                                                             
8000022c:       ddc80813                addi    a6,a6,-548 # 80004004 <test_fp>                                    
80000230:       00004797                auipc   a5,0x4                                                             
80000234:       68078793                addi    a5,a5,1664 # 800048b0 <begin_signature>                            

80000238 <inst_0>:
80000238:       00082a87                flw     fs5,0(a6)                                                          
8000023c:       00482a87                flw     fs5,4(a6)                                                          
80000240:       00205073                fsrmi   zero,0                                                             
80000244:       015afad3                fadd.s  fs5,fs5,fs5                                                        
80000248:       001028f3                frflags a7                                                                 
8000024c:       0157a027                fsw     fs5,0(a5)                                                          
80000250:       0117a223                sw      a7,4(a5)
```

At the same time in priv spec reset section (https://github.com/riscv/riscv-isa-manual/blob/draft-20180709-4dc3fa2/src/machine.tex)
is written that all csrs except couple of fields in mstatus and mcause have undefined state after reset:
```
\section{Reset}
\label{sec:reset}

Upon reset, a hart's privilege mode is set to M.  The {\tt mstatus} fields MIE
and MPRV are reset to 0.  The {\tt pc} is
set to an implementation-defined reset vector.  The {\tt mcause} register is
set to a value indicating the cause of the reset.  All other hart state is
undefined.
```

I think it is should be set to zero state before the first test and zeroed after each next one.

Best regards,
Fedor

neelgala, 2021-11-23:
I agree that the frflags should be set to zero at the start of the test to account for the variable reset state or because of changes incurred within the RVMODEL_BOOT.

I am also inclined to clear our the frflags at the end of each instance, but then we will need separate tests fo checking the stickiness of those flags. 

I'll let @allenjbaum decide on the way forward. This change however will not require changes to the coverage files only the tests and the reference files will have to change. So should be relatively shorter turn-arounds.

fborisovskii, 2021-11-23:
About flags stickness. I thought about it. May be it could be separated test that check stickness or otherwise separated test to check each case of underflow/overflow.... with particular arithmetic instruction. So my comment about zeroing flags after each test is a bit thick. Any way, the initial issue is in macro:
```
#define RVTEST_FP_ENABLE()\
    LI x2, MSTATUS_FS;\
    csrrs x3, mstatus,x0;\
    or x2, x3, x2;\
    csrrw x0,mstatus,x2;
 ```
 The same macro in https://github.com/riscv/riscv-test-env/blob/0666378f353599d01fc48562b431b1dd049faab5/p/riscv_test.h
 has zeroing of fcsr.
 ```
 #define RVTEST_FP_ENABLE                                                \
  li a0, MSTATUS_FS & (MSTATUS_FS >> 1);                                \
  csrs mstatus, a0;                                                     \
  csrwi fcsr, 0
  ```
  So I guess it needs to be added to RVTEST_FP_ENABLE in arch-tech.h if there is no other considerations.

allenjbaum, 2021-11-23:
Tricky.
If we don't clear it before every test, then we are testing that bits are
sticky and sometimes they get properly set, and also they don't get cleared
on their own.
If we do     clear it before every test, then we are testing that bits get
properly set -but not that they're sticky.
I strongly believe that testing that they get properly set should be what
the tests do, so we should add a clear function before each test case.
 (adding a clear function to one of the required macros and to the macro(s)
that store FCSR would minimize source changes)

There should be a separate sticky test (e.g. start with all1s, perform
operations that don't set specific bits and see that all stay 1).
That's a pretty short test.

I'm not sure I agree that coverage won't change; we may find our status bit
coverage gets better.
If we aren't testing fcsr coverage, we must add it. It will be pretty
trivial, just showing that bits transition 0->0, 0->1, 1->1, but not 0->1
(except for the explict clearing)

PR #224 (2021-12-06): Fix RVTEST_FP_ENABLE macro
Reset fcsr before every testset.
Fixes #223
commit 6597bec7 (2021-12-06): Fix RVTEST_FP_ENABLE macro

Reset fcsr before every testset.
Fixes #223
closed 2021-12-06 by neelgala