riscv/riscv-arch-test #227: Makefile for rv32e_unratified

davidharrishmc, 2022-01-17:
It appears that the riscv-arch-test Makefile is only designed to run for rv32i_m and rv64i_m.  For example, SUITEDIR has i_m hardwired into the name. Will it be extended in the near future to also build rv32e_unratified in a standardized fashion?

allenjbaum, 2022-01-17:
RV32E has not been ratified yet; it certainly will at that point. But, I
don't think we should standardize anything that hasn't been ratified.
We may want to talk about what we can do to make unratified tests more
accessible.

davidharrishmc, 2022-02-04:
I built the E tests.  There are a number of issues to report.

1) The signature.output and reference_output files differ by 1-3 final 00000000 entries for most of the tests.  I had to manually change the reference to match the signature.

2) To get the E tests to compile, I had to make the following changes:

Add #define RVTEST_E to the start of each file so it doesn’t try to initialize registers x16-x31
Modify riscv-isa-sim/arch_test_target/spike/model_test.h “sw x1, tohost , t5” to “sw x1, tohost , t1” to use a RV32E register for the HTIF write.

3) The whole process was a major pain because of the way that the makefiles tie the riscv-arch-test directories to the riscv-isa-sim directories.  I had to make some substantial hacks. 

David Harris

neelgala, 2022-02-04:
For issue1: signatures must start and end at a 16-byte boundary [as per spec](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#36-the-test-signature). Those unwanted unwanted regions need to be initialized with 0s. To fix this make sure to add the `.align 4` before the begin/end signature labels as is done here [in spike](https://github.com/riscv-software-src/riscv-isa-sim/blob/master/arch_test_target/spike/model_test.h#L32-L36)

For issue2: I think this has to be fixed and probably the best solution is to add -DRVTEST_E at the end of [this line](https://github.com/riscv-non-isa/riscv-arch-test/blob/307c77b26e070ae85ffea665ad9b642b40e33c86/riscv-test-suite/rv32e_unratified/E/Makefile#L3) and all other Makefile withins the rv32e_unratified director. If you can try this fix and let me know it works, I'll put up a quick PR.

For issue3: Are you using the latest commit of both the repos ? If you clone both the repos (arch-test and spike) and change theTARGETDIR [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/Makefile.include#L2) to point to `<some-path>/riscv-isa-sim/arch_test_target` it should have worked - given issue2 above has been fixed. We have tried to keep the 2 as decoupled as possible. IT would be great if you could point out exactly where the pain-points were so we could improve the framework.

davidharrishmc, 2022-02-09:
Thank you for 1 and 2.

For 3, the riscv-arch-test compiling depends on riscv-isa-sim/arch_test_target/spike/device/device/rv32i_m/I, etc.

This makes compiling new tests difficult.  There's no rv32i_m/F or rv64i_m/D, and no rv32e_unratified.  I had to manually patch things up to match enough to compile.

I pulled the latest commits of both repos today and it seemed to have the same issues.

allenjbaum, 2022-02-09:
For #1: I thought we had that restriction removed; it was originally there
for Spike, but I thought Spike removed that restriction (at our
instigation).

neelgala, 2022-02-09:
@allenjbaum not its not resolved #190 is where its being tracked.

davidharrishmc, 2022-02-09:
One more issue we encountered is that the model_test.h file uses t5 for a write_tohost.  To support RV32E, which lacks t5, I changed it to t1.  

Here's a full set of edits I had to make in riscv-isa-sim.  

$ cd ../arch_test_target/spike/device
$ cp -r rv32i_m/I rv32i_m/F
$ cp -r rv64i_m/I rv64i_m/D
$ sed -i 's/--isa=rv32i/--isa=rv32g/' rv32i_m/F/Makefile.include
$ sed -i 's/--isa=rv64i/--isa=rv64g/' rv64i_m/D/Makefile.include
$ sed -i 's/tohost, t5/tohost , t1/' ../model_test.h

I'm not sure I understand the benefits of tying the riscv-arch-test build directories to mirrored directories in riscv-isa-sim over simply building everything that exists within the riscv-arch-test/riscv-test-suite (and having local directories specifying the ISA flags for simulation). The Makefiles are so abstract and interrelated in their current implementation that they took me a long time to understand.

neelgala, 2022-02-24:
The following PRs address all comments in this issue:
- https://github.com/riscv-software-src/riscv-isa-sim/pull/929
- https://github.com/riscv-non-isa/riscv-arch-test/pull/234

allenjbaum, 2022-03-30:
David: if this fies the problems, could you close this?

closed 2022-03-30 by davidharrishmc