riscv/riscv-arch-test #230: comments in reference_output

davidharrishmc, 2022-02-08:
As we are writing custom tests by hand, it's challenging to keep track of the meaning of the values in the reference_output.  We decided to add comments delimited by # in the reference_output.  We modified one line of the verify.sh script to ignore these comments when running a diff against signature.output, and think this feature might be helpful to other people.

verify.sh
    diff --ignore-case --ignore-trailing-space --strip-trailing-cr <(grep -o '^[^#]*' ${ref}) ${sig} &> /dev/null

neelgala, 2022-02-09:
Well I don't see a harm in adding this as long as its backwards compatible - signatures without # should also match correctly.

just curious.. how are you adding comments to the reference_output - through your own simulator ? Could highlight what type of comments ? Maybe we can get SAIL/Spike to add something similar?

allenjbaum, 2022-02-09:
Yea, I was thinking about that as well. The obvious choice for some
automatic thing might be testcase name, though I'm not sure how to do it.
That would be fantastic if we could, and we could also add it to signatures
written by the trap handler - it would solve a huge problem with
nondeterminstic results.

pawks, 2022-02-09:
> Yea, I was thinking about that as well. The obvious choice for some automatic thing might be testcase name, though I'm not sure how to do it. That would be fantastic if we could, and we could also [...]

One way to do it would be to have a table of strings in the data section or a separate test pointer which increments after each test case. You can store that in the signature section, but that will increase the number of signature words you have to dump out. There are other ways of doing it where you run a post script and use the context from the logs and disassembly to figure out which test caused the write to a particular signature word. Its hard to have a implementation agnostic solution otherwise.

neelgala, 2022-02-09:
@davidharrishmc If I understand correctly the comments have been added manually by Kip in the reference files - there is no tool/automation which goes an updates the reference file automatically ?

@pawks @allen a yet another option : since a lot of simulators (include spike) have a elf-parser, maybe printing the closest-previous label when a signature update happens might work..

pawks, 2022-02-09:
Yes. The information is already present in the elfs, it's a matter of extraction. The only issue with using simulators is that it's simulator/model dependent ( I believe it will be very difficult to achieve this in SAIL). If it is really important, I would recommend having a post processing script which adds these comments later on. That way it's simulator and model agnostic.

davidharrishmc, 2022-02-09:
Neel - correct.  In our use case, both the test cases and the reference_output are written manually, and the comments make it much easier to relate the reference_output to the test cases and help identify where to put more signature entries when more test cases are added by hand.

Automation wouldn’t make sense in this case because there’s not a trivial relationship between tests and signature - the information we want to record on a page fault is different than what we want to record for a successful memory read, and we record no signature for a memory write until we verify it with a read later.

If we were writing an automated test case generator, we would generate comments in the reference_output at the same time we generate the tests and the expected values.

David

neelgala, 2022-02-09:
@pawks maybe the data-propagation module within isac could achieve this.. since it includes an elf-parser and is aware when signature updates happen

neelgala, 2022-02-09:
@davidharrishmc the automation I am refering is purely for the comments in the reference_outputs - which can be independent of the interpretation of the signature contents. The idea is if someone sees a mismatch in the signatures, the comments must help them figure out where in the test they should be focusing on to replicate/isolate the bug.

Since majority of the tests in this repo are automatically generated, we would like to have a similar automation for "comments in reference-signature" as well. Will need to explore more along those lines

neelgala, 2022-02-09:
as a TODO item I will update verify.sh with your command above in my next PR.

closed 2022-02-28 by neelgala

kuopinghsu, 2022-04-21:
@neelgala The BSD diff does not support --ignore-trailing-space option, so it will fail on MacOS. Can it be replaced with --ignore-all-space? It supports both BSD and Linux.

neelgala, 2022-04-21:
if that's the case.. @kuopinghsu would you mind creating a PR for the same ? I can then test it out for linux at my end atleast check for any odd behavior

kuopinghsu, 2022-04-21:
Done. PR #252

PR #234 (2022-02-28): Misc fixes
 - modified verify.sh to ignore comments in reference signature during diff operation [close #230]
 - udpated test-format spec to include the order of lines in the signature file [close #214]
 - RVTEST_E macro to be enabled for all rv32E tests. [#227]
 - Added a disclaimer in README about migration to RISCOF
commit 0ad95eb5 (2022-02-28): ignore comments in reference signature during diff

This should close #230