riscv/riscv-arch-test #232: Unexpected PASS[OK] on RV32I that does not support misaligned instruction addresses

aignacio, 2022-02-20:
Hey everyone,

I have been testing a RV32I CPU that does **not support misaligned instruction addresses** through the **privilege** set of tests  and I was expecting to have all tests failing but ebreak/ecall passing, as it's implemented in the design. But for my surprise **misalign1-jalr-01** test was passing as you can see in the image below.
[image]
While reading the objdump I noticed that the instruction which is suppose to generate a synchronous trap / exception is the one at the line 85 once the jump address is not 4-byte / word aligned.
[image]
The problem that I see is that this jump will never generate a trap once LSB needs to be cleared for the jump target address as specified in the unprivileged ISA.
[image]
Thus technically this test seems contradictory once it should not pass if the design does not support misaligned instr. addresses as mentioned in the [Test Disclaimers](https://github.com/riscv-non-isa/riscv-arch-test#test-disclaimers). Am I missing something?

Ps.: Is there a set of tests for the Zicsr?, if so what is the RISCV_DEVICE value?

pawks, 2022-02-20:
@aignacio The test does behave as intended. This test is used to generate effective addresses which have the LSB set. As pointed out by you in the spec, the LSB should be cleared while generating the target address using JALR. By setting the LSB of the address in the register to 1 we ensure that this operation is performed correctly in the DUT. This test should not generate a trap incase the design conforms to the RISC-V spec. It would be a problem if the test indeed trapped. The `misalign2-jalr-01` is the tests which tests for 2 byte offsets. That test is intended to trap in implementations which do not have the C extension. I think the disclaimer should remove the `misalign1-jalr` from the list of tests which will fail.

aignacio, 2022-02-20:
Thanks for the feedback @pawks, I see your point, so the test disclaimer **needs to be** updated just to avoid the same question again =)

allenjbaum, 2022-02-21:
Just to make this clear: where is this disclaimer that should be modified?

pawks, 2022-02-21:
Its [this](https://github.com/riscv-non-isa/riscv-arch-test/blob/307c77b26e070ae85ffea665ad9b642b40e33c86/README.md?plain=1#L33) line here which needs to be changed to:
```
   2. rv[32/64]i_m/privilege/src/misalign2-jalr-01.S
```

allenjbaum, 2022-02-21:
got it, thanks

closed 2022-02-22 by aignacio

aignacio, 2022-03-07:
@allenjbaum are you going to update the test disclaimers?

allenjbaum, 2022-03-08:
I was actually going to have the person that added the disclaimer do that.
I will contact him.

pawks, 2022-03-09:
Done [here](https://github.com/riscv-non-isa/riscv-arch-test/pull/231/commits/bbaa7b66bacc62b6d8b3a29c148cc90fb5ca5e6f).