riscv/riscv-arch-test #233: Update isa for privilege tests.

pawks, 2022-02-23:
The latest version of the toolchain expects `_Zicsr` to be present in the `march` argument for tests which execute `csr*` instructions. The current tests do not include this extension in the argument to the `RVTEST_ISA` macro. Any test which needs to include the trap handler (via the `rvtest_mtrap_routine`) conditionally or unconditionally will have to be updated according to the definition [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/riscof-dev/spec/TestFormatSpec.adoc#431-required-pre-defined-macros).

allenjbaum, 2022-02-23:
I'm not sure which definition you're pointing to in the testformatspec doc.

I think I need to modify my trap handler to make sure that works correctly.

pdonahue-ventana, 2022-02-23:
Is this related to https://github.com/riscv-software-src/riscof/issues/35?

allenjbaum, 2022-02-23:
> I was pointing to the RVTEST_ISA macro. I don't think the trap handler
> needs to be updated. Only the isa strings need to be updated.

I misstated; it's actually the RVTEST_CODE_END that supports the (new) trap
handler that needs to change. It should be a simple fix.

pawks, 2022-02-23:
@pdonahue-ventana Yes.

allenjbaum, 2022-03-30:
Could this be fixed by defining changing 
#define  RVTEST_ISA(_STR)    to
#ifdef rvtest_mtrap_routine
  #define  RVTEST_ISA(_STR##_Zicsr)
#else
  #define  RVTEST_ISA(_STR)
#endif

 so we only change it in one place rather than many? (assuming I have my syntax correct)

pawks, 2022-03-30:
> Could this be fixed by defining changing #define RVTEST_ISA(_STR) to #ifdef rvtest_mtrap_routine #define RVTEST_ISA(_STR##_Zicsr) #else #define RVTEST_ISA(_STR) #endif
> 
> so we only change it in one place rather than many? (assuming I have my syntax correct)

That would not work because riscof parses the files to get the ISA and the macro does not really mean anything in the code. The macro is an empty macro.

pawks, 2022-03-30:
Another solution would be to have a policy to include `zicsr` in the march string always(if it is present in the isa of input yaml). This would also allow the boot code to contain csr instructions. That way the tests may or may not have the `Zicsr` explicitly set in the `RVTEST_ISA` string.

allenjbaum, 2022-03-30:
That's too simple; there must be a downside, e.g. it would cause some CSR
tests on machines that don't implement it?

pawks, 2022-03-30:
> That's too simple; there must be a downside, e.g. it would cause some CSR tests on machines that don't implement it?
> […](#)

Not really. The `check` statements in the `RVTEST_CASE` macro drive the conditions for choosing a test. The `RVTEST_ISA` macro is just a indication of the string which should be passed to the `march` argument of gcc while compiling. Adding `Zicsr` will not affect the test filtering in any way.

allenjbaum, 2022-03-30:
As long as it only affects tools, then sure. Which file needs to change for
this to happen?

allenjbaum, 2022-08-25:
Close after documenting that plugin should add or omit  -Zicsr depending on toolchain. 
Close after updating all priv tests to include Zicsr.

pawks, 2022-11-04:
Closed with #286

closed 2022-11-04 by pawks