riscv/riscv-arch-test #235: Floating point test size

davidharrishmc, 2022-02-27:
The rv32f fmadd_b15, fmsub_b15-01, fnmadd-b15-01, and fnmsub-b15-01 tests are an order of magnitude larger than any others in the test suite.  They take a long time to compile and simulate.  The b15 tests are omitted from rv64d, presumably because they are too big, even though they are in the riscv-ctg.  

I've had trouble finding where the 'ibm_b15 cases are defined. Does anyone have insight into whether the tests are worth including, or could be shrunk in some graceful way?

neelgala, 2022-02-27:
ibm_b15 abstract functions : https://riscv-isac.readthedocs.io/en/stable/code.html#riscv_isac.fp_dataset.ibm_b15

allenjbaum, 2022-03-30:
IF this answers your question, could you close this? 
This is the hardest FP function to test, MFA, so it's a cross product between both add and multiply
I don't know if/when those tests will be added to RV64D (and therefore RV32D).