riscv/riscv-arch-test #239: All privilege tests must define rvtest_mtrap_routine

allenjbaum, 2022-03-15:
Any tests that can possibly trap (and not immediately hang) must install a trap handler.
This is done by defining the variable  rvtest_mtrap_routine (and, if it traps into smode, also 
 rvtest_strap_routine, and if it traps into a guest OS,  rvtest_vtrap_routine).

None of the priv tests currently define this variable, so the trap handler code defaults to defining it.
There might b a better way than explicitly defining it in every test, but I don't see any obvious way to do that.

neelgala, 2022-03-15:
All privilege indeed define the rvtest_mtrap_routine macro here : https://github.com/riscv-non-isa/riscv-arch-test/blob/5932678c2edac2606f366144a1ed7ce5cc3cd611/riscv-test-suite/rv32i_m/privilege/Makefile#L3

allenjbaum, 2022-03-15:
Ah,that's where it is. Then we have a different problem. Thanks.