riscv/riscv-arch-test #240: Edge signatures are 0xdeadbeef now

marcfedorow, 2022-03-16:
Since this patch every test that has >= 2048 bytes of signature that edge block fails.
This works if you change last signature block address (2044) to zero (0).
But I thins that this should be fixed somehow different.
Reverting "[Automatically adjust base and offset if offset gets too big](https://github.com/riscv-non-isa/riscv-arch-test/commit/49e3e0b2efcbf3cda315d5faf6d3dfb82fc0f505)" works for me.

allenjbaum, 2022-03-16:
Could you send the name of the failing test and the signature that
mismatched? Obviously that's not supposed to be happening....

marcfedorow, 2022-03-16:
It is common for all the tests with long enough signature.Just for example let’s tale rv32i andi test. There is the issue:0848 < 000000020848 > deadbeef0848   ^^^^^^^^ There is the code:inst_530:// rs1_val==6 and imm_val==819, // opcode: andi ; op1:x10; dest:x11; op1val:0x6;  immval:0x333TEST_IMM_OP( andi, x11, x10, 0x2, 0x6, 0x333, x2, 2044, x6)RVTEST_SIGBASE( x2,signature_x2_1) От: Allen BaumОтправлено: 16 марта 2022 г. в 23:10Кому: riscv-non-isa/riscv-arch-testКопия: marcfedorow; AuthorТема: Re: [riscv-non-isa/riscv-arch-test] Edge signatures are 0xdeadbeef now (Issue #240) Could you send the name of the failing test and the signature thatmismatched? Obviously that's not supposed to be happening....

marcfedorow, 2022-03-16:
It is common for all the tests with long enough signature.Just for example let’s tale rv32i andi test.

There is the issue:
```
0848 < 000000020
848 > deadbeef0
848   ^^^^^^^^
```

There is the code:
```
Inst_530:
// rs1_val==6 and imm_val==819,
// opcode: andi ; op1:x10; dest:x11; op1val:0x6;  immval:0x333
TEST_IMM_OP( andi, x11, x10, 0x2, 0x6, 0x333, x2, 2044, x6)
RVTEST_SIGBASE( x2,signature_x2_1)
```

allenjbaum, 2022-03-16:
I think the problem is that this code will work with the next version of
the framework (which we will cut over to in May)
but not the current one when the explicit offset is 2044 (and I tested the
code with the new framework
I have a fix...

Imperas, 2022-03-16:
Allen - I don't understand - the tests and the generated signatures are a function of the programs/assembler/macros - and have nothing to do with the script or python that run the RTL or simulators - so can you explain actually what is wrong with this Makefile/Bash framework that will be sorted in a python one - as the programs still are assembled with the same tools, and run on the same RTL or models,  and the signatures are written by the same programs, and the signatures are dumped by the RTL and simulators - so what am I missing... thnx - Simon

allenjbaum, 2022-03-16:
I don't understand your question. This is something that was intended to be
used in a riscof framework and tested there.
It has absolutely nothing to do with python, it has to do with introducing
it without testing it on the existing tests on the old framework (though I
think it was tested on new tests going into the old framework).
So, left hand, right hand.

marcfedorow, 2022-03-16:
Can you please revert a patch that introduced this bug before next version of framework is published?

closed 2022-03-18 by neelgala

allenjbaum, 2022-10-11:
Either revert or fix, but yes, we will

commit f8727ccf (2022-03-17): Patch for issue 240. Fixes #240
commit 0f6f4089 (2022-05-04): Merge pull request #242 from riscv-non-isa/fix-offset-bug

Patch for issue 240. Fixes #240