riscv/riscv-arch-test #241: Deprecated macro warnings

marcfedorow, 2022-03-16:
There are lots of warnings that some macro is deprecated and some other macro should be used.
I compile these tests using default options mostly.
What options should I pass so that not deprecated macros will be used?

pawks, 2022-03-17:
@marcfedorow The deprecated macros refer to the model specific `RVTEST_*` macros which are defined by the implementer in `model_test.h`. To enable easier migration the older macros have been used to define the latest macros. It is advised that you rename those macros in your model specific files as they will not be supported in the future. The list of model specific macros and their intended operation can be found [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/master/spec/TestFormatSpec.adoc#432-required-model-defined-macros). You can find the deprecated macros and how they are redefined as per the new spec over [here](https://github.com/riscv-non-isa/riscv-arch-test/blob/5932678c2edac2606f366144a1ed7ce5cc3cd611/riscv-test-suite/env/arch_test.h#L1313).

allenjbaum, 2022-03-30:
IF this answers your question, please close the issue. If it doesn't, please explain what is missing.

closed 2022-03-30 by marcfedorow