riscv/riscv-arch-test #249: RISC-V Toolchain

duncangraham-Imperas, 2022-04-08:
I attempted to run the lastest master branch but with the changes for the ratified crypto (K) tests I now get assembler errors for the xperm8 instruction.
Could you recommend a source of a binary GNU GCC cross-compiler toolchain that satisifes these tests?

allenjbaum, 2022-04-11:
Could this because xperm is in the Zbkx extension, which makes it look like
it's a bitmanip op - except it was never a ratified?
You may have to explicitly list Zbkx in the march string (this is a guess)

allenjbaum, 2022-04-22:
Have you tried this fix?

allenjbaum, 2022-04-22:
also: have you tried running this under riscof?

Imperas, 2022-04-25:
@allenjbaum does riscof have its own tool chains? surely not. 
why would running under riscof change the toolchain installation.

can you pls get somebody to answer Duncan's question 'Could you recommend a source of a binary GNU GCC cross-compiler toolchain that satisifes these tests?'
thnx
Simon

allenjbaum, 2022-04-27:
As far as I know we have people using it all the time with these tests. so I don't know what the problem is.
i'll see what I can find out.

Imperas, 2022-04-27:
Thanks @pawks  for this info - can you respond with the location to download a toolchain that accepts this option. thanks. Simon

allenjbaum, 2022-04-27:
Can I assume that you've applied the fix to the makefile, and it doesn't work on the version of the toolchain that previously failed?

duncangraham-Imperas, 2022-04-27:
Yes, replaced _rv64ib_ with _rv64iZbkx_ in _riscv-test-suite/rv64i_m/K/Makefile_. With the toolchain I have previously used successfully for these tests (and when also trying with others) I get the error :
  _riscv64-unknown-elf-gcc: error: '-march=rv64iZbkx': unsupported ISA subset 'Z'_
Thanks for your help

neelgala, 2022-05-26:
tests currently checked in will require `RVTEST_ISA` macro updates to get the dut-plugins notified of the extensions required to compile these tests.

allenjbaum, 2023-02-09:
Duncan, I haven't seen this closed or a note that says it is still broken. I'm going to close this and you can reopen it if it isn't fixed

closed 2023-02-09 by allenjbaum