riscv/riscv-arch-test #250: Possible bugs in D test macros

misterjdrg, 2022-04-16:
So, I looking onto generated assembly for fadd.d and seeing this:
```
00000000800004ac <inst_0>:
   800004ac:   00083d87                fld     fs11,0(a6)
   800004b0:   00883d87                fld     fs11,8(a6)
   800004b4:   00205073                csrrwi  zero,frm,0
   800004b8:   03bdfdd3                fadd.d  fs11,fs11,fs11
   800004bc:   001028f3                csrrs   a7,fflags,zero
   800004c0:   01b7b027                fsd     fs11,0(a5)
   800004c4:   0117b423                sd      a7,8(a5)

00000000800004c8 <inst_1>:
   800004c8:   01083587                fld     fa1,16(a6)
   800004cc:   01883487                fld     fs1,24(a6)
   800004d0:   00205073                csrrwi  zero,frm,0
   800004d4:   0295f4d3                fadd.d  fs1,fa1,fs1
   800004d8:   001028f3                csrrs   a7,fflags,zero
   800004dc:   0097b827                fsd     fs1,16(a5)
   800004e0:   0117bc23                sd      a7,24(a5)
```
So csrrwi@800004d0 cleaning only rm part of fcsr.

And fflags stay clobbered for the rest of testcases?

commit: 6d87f30 (current master)

pawks, 2022-04-16:
The instruction at `800004d0` is setting the `frm` field of the csr for the operation and not clearing it. That being said, it does appear that the `fflags` is never cleared and it stays clobbered for the rest of the cases(`csrrs` does not write the csr when `rs1` is `x0`). The simplest fix would be to change the `csrrs` after the `fadd.d` to `csrrw`.

misterjdrg, 2022-04-16:
So, if I understand correctly 2 lines needs to be changed right here
https://github.com/riscv-non-isa/riscv-arch-test/blob/6d87f30f0a1166088d7a007fca69fd4d1de12410/riscv-test-suite/env/arch_test.h#L926

misterjdrg, 2022-04-16:
Patch is done.
How I can regenerate signatures in `riscv-test-suite` ?

pawks, 2022-04-16:
All the F* macros will need to be updated. I believe there are 8 of them. You can run the tests on the SAIL model and generate the signatures for each of them. You will however have to replace the signature files in the `riscv-test-suite/rv[32/64]i_m/[I/D]/references` directories. You can use riscof or the framework in this repository to generate them.

misterjdrg, 2022-04-16:
I'm running tests like that
`make RISCV_PREFIX=riscv64-elf- RISCV_TARGET=spike_0p1 RISCV_DEVICE=D XLEN=64 TARGET_SIM="spike" verify`
I tested only `XLEN=64`.
Only D tests are failing.
So I can just copy signatures from `work/rv64i_m/D/*.signature.output` into `riscv-test-suite/rv64i_m/D/references/*.reference_output ` ?

allenjbaum, 2022-04-17:
If the flags aren't being reset between tests, and the coverage report
didn't show anything missing, it sounds like we have a hole in our coverage
goals.
IF the coverage is measure whether a bit is set, then it's really measuring
that it either was set, or is getting set - and we need to separate them
out.
So any tests that doesn't set it will be have a signature match - but can't
tell whether it mistaken set status where the bit had been previously set.
We probably want some tests that show that status bits never spontaneously
get cleared, and can only explicitly be cleared.

pawks, 2022-04-17:
> I'm running tests like that `make RISCV_PREFIX=riscv64-elf- RISCV_TARGET=spike_0p1 RISCV_DEVICE=D XLEN=64 TARGET_SIM="spike" verify` I tested only `XLEN=64`. Only D tests are failing. So I can just [...]
Yes. That should be enough.

> If the flags aren't being reset between tests, and the coverage report didn't show anything missing, it sounds like we have a hole in our coverage goals. IF the coverage is measure whether a bit is [...]

The problem is not with the coverpoints because the flags aren't a part of the coverpoints at all, not currently at least. Our coverpoints are defined only on the input values of the instructions and not on the side-effects or the outputs i.e to see if the divide by 0 flag is set, the rs2 val is set to 0. The coverage metrics do not have any conditions defined over the flags. We might want to add csr-state info as a part of the coverpoints so that we can capture coverpoints which test stickiness of the bits.

allenjbaum, 2022-04-18:
If the tests store the status bits in the signature (they do; that's the whole poont of the SIPD_FID, then there should be a coverpoint for it - otherwise what is the point (sic)?
Let's get those added.

pawks, 2022-04-19:
Reopening this to track cgf updates capturing coverpoints dependent on `fflags`.

eroom1966, 2022-04-19:
This whole issue described above is about causality, not coverage,

> causality - the relationship between cause and effect.

This can be simplified down to something far less complicated, for example

```
// pseudo-code
mv r1, 2
mv r2, 3
mv r3, 8
add r31, r1, r2 // r31 -> (2 + 3) = 5
SAVE_SIG(r31) // save 5
sub r31, r3, r2 // r31 -> (8 - 3) = 5
SAVE_SIG(r31) // save 5
```

Does this test prove that the **sub** instruction worked correctly ?
no it does not, if I were to remove the sub instruction completely and replace with a **nop**, the test would yield the same signature values
If I were to enhance the test to do the following

```
// pseudo-code
mv r1, 2
mv r2, 3
mv r3, 8
add r31, r1, r2 // r31 -> (2 + 3) = 5
SAVE_SIG(r31) // save 5
mv r31, 0
SAVE_SIG(r31) // save 0
sub r31, r3, r2 // r31 -> (8 - 3) = 5
SAVE_SIG(r31) // save 5
```
This would catch an ineffective **sub** instruction, as the final saved value would remain at zero 

How can you catch these cases whereby there are errors or misinterpretations in causality ?
Mutating a test, and asserting that the results change is a way to catch these coding errors.
If you mutate a test, and the result produced is the same, then the test s not capturing what you thought it was doing 

@allenjbaum I agree with your solution, that the flags need to be reset as they are sticky
The issue, is that this had to be discovered by manual interrogation of the test by @misterjdrg rather than automation being available to direct you to the hole in the test, when the test was created
That is what mutation testing would have told you

Thx
Lee

allenjbaum, 2022-04-20:
Our testing methodology doesn't (or hasn't) produced tests like that, so
that particular example is a non issue.
And, frankly it's pretty easy to ensure that tests generate unique values
instead of the same one over and over,
and that they don't continue to target the same destination. The coverage
metrics we have tend to discourage that as a side effect, in fact.
Regardless of that, it would have to have incorrect behavior on this masked
case, but correct behavior on every other SUB instruction test
  (since we are not expected to see transient failures) to be completely
missed.
This is another case of DV vs. compatibility testing differences.

What we are testing is that "implementors have read, understood, and
implemented the spec"-- and not that it completely bug free, which is what
DV is looking for.
If they haven't understood the spec, or understood it but decided to take a
shortcut  because it was easier to implement that way - that's what we are
trying to test.
Yes, the chip could have a bug such that some combination of registers and
ops failed to write the destination register.
If our tests only test a tiny number of those ops, we might not catch it.
We don't do that, so it's more than highly likely we would catch a bug like
that.
And, frankly, that case is  a rather over-the-top bug, not a mere
inconmpatibility. Not our goal.
We may not catch all of the bugs, but it doesn't matter; catching just one
is good enough to show incompatibility, and that is what our goal is.
So, mutation is nice, and if you donated it to RVI we might use it. IF it
causes test generation time to explode, we might not, as well.
But in any case, missing some specific case while catching all the others:
not an issue.

I maintain that coverage is exactly the issue here. Causality is nice, a
totally different technique that we don't have access tp
(or budget to create),
(or budget to license)
so we will make do with open source tools that we don't have to license,
whose source code we have access to,
and that we can modify and fix easily (which, in this case, is already
partially implemented).

pawks, 2022-09-22:
I believe this has been fixed by PRs #274 #271 #265.

allenjbaum, 2022-12-08:
This now needs new coverpoints, which may require new tests, but that will be dealt with from the ctg side to generate additional tests

commit 96e3589a (2022-04-18): Merge pull request #251 from misterjdrg/clean_fflags. Closes #250

Cleaning fflags csr after each test
closed 2022-12-08 by allenjbaum