riscv/riscv-arch-test #253: Possible bugs in D test macros v2

misterjdrg, 2022-04-30:
So, I looking onto generated assembly for `fcvt.d.s` and seeing this:

```
00000000800004ac <inst_0>:
    800004ac:   00083c87                fld     fs9,0(a6)
    800004b0:   00205073                fsrmi   zero,0
    800004b4:   420c86d3                fcvt.d.s        fa3,fs9
    800004b8:   001018f3                fsflags a7,zero
    800004bc:   00d7b027                fsd     fa3,0(a5)
    800004c0:   0117b423                sd      a7,8(a5)
```

`fcvt.d.s` is converting from f32 to f64.
So `fs9` @`800004b4` should contain 32bit float.
`fld` loading f64 into `fs9`
Maybe `flw` should be used?

allenjbaum, 2022-05-01:
IS this on an RV64 or an RV32?
For an RV64, I think we need to see that this works properly with a
NaN-boxed value, so it loads all 64 bits.
For a single precision float, this will look like a NaN, but the cvt.d.s
will notice that the upper 32bits are all 1 and will treat it as a single
instead of an actual NaN.

misterjdrg, 2022-05-01:
This is RV64.
Values in tests doesn't look like 64bit floats and most results of convertion is NaN.
I trying to generate unit tests suite for my emulator, so
I patched spike to generate detailed trac
[d_fcvt.d.s_b22-01.spike.reglog.txt](https://github.com/riscv-non-isa/riscv-arch-test/files/8599666/d_fcvt.d.s_b22-01.spike.reglog.txt)
es of execution.

allenjbaum, 2022-05-02:
If you do a fld of a properly Nan-boxed single precision value (so,
32bits of all1, 32 bits 0f fp single)
or an.       flw of a single precision value,
and then convert single->double and then fsd, you should not be getting a
Nan result unless the 32 bit of FP single was a Nan
If you did a fld of a single precision value where the upper 32bits were
not all1), you'll get a Nan upon conversion.
That's a valid test to show that nan-boing is working properly.

So,
 - are you using fld or flw, and
 - what is the value in hex that you're loading, and
 -what is the value being stored?

misterjdrg, 2022-05-02:
Form riscv spec:
> When multiple floating-point precisions are supported, then valid values of narrower n-bit types,
n < FLEN, are represented in the lower n bits of an FLEN-bit NaN value, in a process termed
NaN-boxing. The upper bits of a valid NaN-boxed value must be all 1s.

So if 32 bit is properly boxed, all bits above 32-th should be 1
This is not the case

allenjbaum, 2022-05-02:
If I am understanding you correctly (not a certainty), you are saying that
some tests are doing a 64bit fld instruction of a value that doesn't not
have the upper bits all1,
and that the result of converting it to a double is a Nan.

As I explained, that is the correct result, and a perfectly valid test to
demonstrate that an implementation is checking that the upper bits are all1.

What assumption am I making here that is not correct?

misterjdrg, 2022-05-07:
It true.
I mean `fcvt.d.s` doesn't have a single 'happy path' test, where 127-32 are ones.

neelgala, 2022-05-07:
The issues mentioned here are already captured in [this](https://github.com/riscv-software-src/riscv-isac/issues/38) and [this](https://github.com/riscv-software-src/riscv-isac/issues/39) issue on isac and are being worked on.

misterjdrg, 2022-05-07:
Ok, thanks

closed 2022-05-07 by misterjdrg