riscv/riscv-arch-test #256: some float/double cases are not right

zhongxixuzhi, 2022-05-09:
for example: 
d_fdiv_b1_01.S
inst_13:
// rs1==f0, rs2==f5, rd==f17, fs1 == 1 and fe1 == 0x3f8 and fm1 == 0x0000000000000 and fs2 == 1 and fe2 == 0x001 and fm2 == 0x0000000000002 and rm_val == 0  
// opcode: fdiv.d ; op1:f0; op2:f5; dest:f17; op1val:0xbf80000000000000; op2val:0x8010000000000002; valaddr_reg:x16; val_offset:208; rmval:0x0; correctval:0; testreg:x18
TEST_FPRR_OP(fdiv.d, f17, f0, f5, 0x0, 0, x16, 208, x17, x15, 208, x18)

0xbf80000000000000 divide 0x8010000000000002, and result is stored into f17, it is not equal 0, but you set the correct val rm_val == 0, so this case is not right.
it seems that there are lots of similar cases in riscv-test-suite/rv64i_m/D.

is there anyone who can help to update? thanks a lot~

allenjbaum, 2022-05-09:
Just to be clear: is it the actual result that is incorrect, or the comment with the result incorrect?
Note that *every* result reported is zero, which make me think this is a test generator artifact.

neelgala, 2022-05-12:
@allenjbaum the comments for the reference-value/correct-value for almost all tests must be ignored because they are not meant to be generated by CTG. They are a place holder for a fix that @MarcKarasek volunteered to fix by extracting those from the SAIL runs and pushing them back into the tests via some compile macros.

So, I would not focus on the "reference-value/correct-val" in the comments for now.

zhongxixuzhi, 2022-05-16:
@allenjbaum  the the actual result is incorrect. @neelgala do you know how to fix it? thanks a lot~~

neelgala, 2022-05-16:
The results generated are from SAIL and/or Spike (default golden models). If you can show that SAIL/Spike (both of which use soft-float) infact spit out the wrong values, then this should be filed with SAIL/Spike ASAP. 

Considering this repo (riscv-arch-test) has migrated to RISCOF and no longer host the references in this repository there is nothing much to be done in the tests themselves.

closed 2022-11-04 by pawks