riscv/riscv-arch-test #261: RISC-V COMPLIANCE on SweRV EL2

KinzaQamar, 2022-06-12:
Hi,
I have integrated a target for SweRV El2 on branch 1.0. The core is an industry-standard core and it is great if we have a target to run compliance on that. 
If you guys are interested, I can send a PR for that.

allenjbaum, 2022-06-22:
Could you point at your proposed PR? I'm unclear exactly what you are proposing. Certainly we would like  to ensure that ACTs can be run easily on the SweRV core!

allenjbaum, 2022-06-23:
> Hi @allenjbaum <https://github.com/allenjbaum> ,
> I haven't sent any PR I'll send that out in few days.
> The core itself has no compliance target. The ACT was integrated on
> SwerVolf which is a EL2 based SoC.

I'm not sure what you mean by target. Now that we are using riscof, the new
version of the ACT framework, the target directories have all been removed.
Do you mean you have defined the proper plugins and other support needed by
riscof?

allenjbaum, 2022-12-08:
No PR has been submitted, so closing this. Reopen if there is something actionable.

closed 2022-12-08 by allenjbaum