riscv/riscv-arch-test #262: Possible bug in priviledge/missaligned load tests

miogui, 2022-06-15:
Hi, 

I'm having some issue with the rv64i_m/privilege/misalign_l* tests, all others test are passing fine.

From what I've seen, on a trap (misaligned  load or store), the common_mexcpt_handler save the offset between mtval and mtrap_sigptr (inside the RVMODEL_DATA section) into the trap signature area.

```       	
   adj_mtval:
          csrr   t2, CSR_MCAUSE  /* code begin adjustment amount already in t3 */
  
          LI(t4, CODE_REL_TVAL_MSK)   /* trap#s 12, 3,1,0, -- adjust w/ code_begin */
          sll     t4, t4, t2		          /* put bit# in MSB */
          bltz    t4, sv_mtval		        /* correct adjustment is code_begin in t3 */
  
          LA(     t3, mtrap_sigptr) /* adjustment assuming access is to signature region */ 
          ```

It works well for the Store misaligned but the TEST_LOAD macro use the symbol rvtest_data as a base address for load, which is inside the RVTEST_DATA section, not the RVMODEL_DATA. This could lead in some issue if we do not use the default sections address. 
```
RVTEST_DATA_BEGIN
.align 4
rvtest_data:
.word 0xbabecafe
RVTEST_DATA_END

RVMODEL_DATA_BEGIN
....
....
#ifdef rvtest_mtrap_routine

mtrap_sigptr:
    .fill 64*(XLEN/32),4,0xdeadbeef

#endif
....
....
RVMODEL_DATA_END
```

Is there any reason to have a base adress set to rvtest_data in TEST_LOAD whilst TEST_STORE use the swreg value?
```
#define TEST_STORE(swreg,testreg,index,rs1,rs2,rs2_val,imm_val,offset,inst,adj)   ;\
LI(rs2,rs2_val)                                                             ;\
addi rs1,swreg,offset+adj                                                     ;\
LI(testreg,imm_val)                                                         ;\
sub rs1,rs1,testreg                                                          ;\
inst rs2, imm_val(rs1)                                                      ;\
nop                                                                         ;\
nop                                                                         

#define TEST_LOAD(swreg,testreg,index,rs1,destreg,imm_val,offset,inst,adj)   ;\
LA(rs1,rvtest_data+(index*4)+adj-imm_val)                                      ;\
inst destreg, imm_val(rs1)                                                   ;\
nop                                                                         ;\
nop                                                                         ;\
RVTEST_SIGUPD(swreg,destreg,offset) 

```

For some PMA/cacheability and testbench reasons, I have to move the signature section at 0x30000000.

miogui, 2022-06-15:
> Store tests use RVMODEL_DATA as a base because they are storing directly
into the signature area.

yes I understand but TEST_STORE does not set the base address directly, and the test is responsible to set the right address with the swreg, testreg and offset/adj/imm, whilst the TEST_LOAD do not use the swreg/testreg to set the base address.

We should be able to move the signature sections without breaking the reference check.

allenjbaum, 2022-06-16:
The trap handler converts xTVAL to an offset from rvtest_data_begin label
if the cause is a data exception (as opposed to a code exception or illegal
op)
I've added code to the next version of the handler so that  on a data
exception,
    if xTVAL is between rvtest_sig_begin..rvtest_sig_end,
    then the offset is relative to rvtest_sig_begin instead  of
rvtest_data_begin
(and defined those labels).
Does that address your issue?

miogui, 2022-06-16:
`The trap handler converts xTVAL to an offset from rvtest_data_begin label
if the cause is a data exception`
I've probably miss something here, from what I see in my trace is that the xTVAL is an offset from mtrap_sigptr which is already inside the rvtest_sig section.

Looking at the the trap handler, I see that there's already the two adjustement but the convertion to an offset from rvtest_data is done only if xTVAL > mtrap_sigptr.
So the solution you give seem's  to address the issue.

allenjbaum, 2022-06-16:
You're looking at the existing trap handler; the new one treats it
differently and there will be a PR for it shortly after we test it a bit
more.

pawks, 2022-10-18:
The closing of this issue was an error. The new trap handler is supposed to fix this.

Abdulwadoodd, 2022-10-18:
> The closing of this issue was an error. The new trap handler is supposed to fix this.

Agreed!

The new trap handler also causes another signature value mismatch between DUT (RTL) and Reference (SAIL) - that is the 1st argument of the trap handler, apparently. As a result, a bunch of other privilege tests (excluding misaligned branch and jump tests) is failing in addition to misaligned load and store.

allenjbaum, 2023-02-09:
The above comment isn't quite enough to be able to tell what went wrong. 
Please retest with the current version of the trap handler and either close this, or indicate the test that failed (even if it's not a test in the repo, in which case send a pointer) and the error report . 
The relocation is now based on which region TVAL is in, rather than whether it was a code or data exception, and extends relocation for EPC as well, which may fix issues with tests that were not writen as expected.

pawks, 2023-02-09:
I've verified that this bug does not exist in the latest release of arch-test. Closing it.

PR #286 (2022-10-18): Clean up
This PR has the following changes.
- Canaries (labels - sig_begin_canary. tsig_begin_canary, tsig_end_canary, sig_end_canary)
- Signature boundary labels (rvtest_sig_begin and rvtest_sig_end) (enables the new trap handler to fix #262)
- Zicsr ISA update for priv tests (#233)
- Default data section should be 16 bytes. This expands default rvtest_data region to be at least 16 bytes (#211)
- Replace la/li ops with LA/LI macros in tests (#275)
- Remove trap handler enable macro from misalign1-jalr tests (#281)
- Move misalign1-jalr test into I directory. (#281)
- Move fmem tests into the  [... cut]
closed 2023-02-09 by pawks