riscv/riscv-arch-test #263: Complete PMP configuration CSRs and address CSRs are not supported

Abdulwadoodd, 2022-06-29:
Arch-tests do not have the support of following CSRs in `encoding.h` file.
- pmpcfg`x` where x is [4,15]
- pmpaddr`x` where x is [16,63]

Abdulwadoodd, 2022-06-29:
Little correction: The file present by default in [riscv/riscv-opcodes](https://github.com/riscv/riscv-opcodes) is not complete but the most up-to-date version (even latest than what is in [riscv-isa-sim/riscv](https://github.com/riscv-software-src/riscv-isa-sim/tree/master/riscv)) of the file can be generated by running `make` in [riscv/riscv-opcodes](https://github.com/riscv/riscv-opcodes).

allenjbaum, 2022-06-30:
I've filed an issue to try to ensure that the version in the repository is always the latest.
The question is how the framework pulls it in, or should the framework always regenerate it as you did
In either case it doesn't need to be in the repo at all.

Abdulwadoodd, 2022-06-30:
According to Andrew's comment on your issue ([here](https://github.com/riscv/riscv-opcodes/issues/132#issuecomment-1170642498)) for now, we need to regenerate it every time. How about writing a makefile that clone [riscv/riscv-opcodes](https://github.com/riscv/riscv-opcodes)[riscv/riscv-opcodes](https://github.com/riscv/riscv-opcodes) and build encoding.h locally to place in [riscv-arch-test/riscv-test-suite/env](https://github.com/riscv-non-isa/riscv-arch-test/tree/main/riscv-test-suite/env). What do you say

allenjbaum, 2022-07-01:
We could either
 - 1. clone it as you said
 - 2. add a CI script to the riscv-opcodes repo to produce it as a
release and simply copy it from as part of make
 - 3.copy it from the spike directory.

Short term, your solution is best.
Longer term: I'd like to see the CI script solution so others can use the
same approach, if that is possible

Abdulwadoodd, 2022-07-03:
Hi @allenjbaum, I have implemented a long-term solution by writing a CI script. Kindly review PR #268

closed 2022-07-21 by Abdulwadoodd