riscv/riscv-arch-test #266: RVTEST_CASE d_fsd-align and d_fld-align regex typo

DTowersM, 2022-06-30:
The following two tests have a typo on line 33. The RVTEST_CASE has a regex that checks that the ISA is valid. The following tests check that the ISA is 32 bit, but they are both 64 bit tests. This causes these two tests to not build in the current riscof flow, as they are filtered out by the regex check. 

/riscv-arch-test/riscv-test-suite/rv64i_m/D/src/d_fsd-align-01.S
/riscv-arch-test/riscv-test-suite/rv64i_m/D/src/d_fld-align-01.S

RVTEST_CASE(0,"//check ISA:=regex(.\*32.\*);check ISA:=regex(.\*I.\*D.\*);def TEST_CASE_1=True;",d_fsd-align)
RVTEST_CASE(0,"//check ISA:=regex(.\*32.\*);check ISA:=regex(.\*I.\*D.\*);def TEST_CASE_1=True;",d_fld-align)

We think these should be:

RVTEST_CASE(0,"//check ISA:=regex(.\***64**.\*);check ISA:=regex(.\*I.\*D.\*);def TEST_CASE_1=True;",d_fsd-align)
RVTEST_CASE(0,"//check ISA:=regex(.\***64**.\*);check ISA:=regex(.\*I.\*D.\*);def TEST_CASE_1=True;",d_fld-align)

Are we missing something?

pawks, 2022-06-30:
These aren't 64 bit only tests. They are applicable for RV32D systems too. The conditions for these tests aren't right, the xlen shouldn't be checked for these particular tests at all. All of the floating point tests are getting replaced soon. Refer to the `fp-tests` branch for an updated version of the dp suite.

ptprasanna, 2022-07-05:
did you meant fp-dev branch? please point the repo path otherwise.

pawks, 2022-07-05:
The PR has been merged and the tests have been updated on main.

allenjbaum, 2022-07-20:
IF this has been merged, can what happened to all the load and store tests? They are no longer in any of the
 rv32i_m/F/src, rv32i_m/D/src, rv64i_m/F/src, and rv64i_m/D/src directories

pawks, 2022-07-20:
Oops. Looks like there was an oversight for the fp memory operations. I will check those in too.

allenjbaum, 2022-09-22:
fixed with PR#274

closed 2022-09-22 by allenjbaum