riscv/riscv-arch-test #272: Macro used in Crypto Scalar code mislabelled

allenjbaum, 2022-07-26:
There was a commit that modified the TEST_RI macro to add a second source register, but didn't change the name.
Effectively, this changed the macro from TEST_RI to TEST_RRI which already existed.
Oddly, TEST_RRI is unused by any test.
The fix is to edit all tests that use TEST_RI to instead use TEST_RRI (since they're currently identical) and delete TEST_RI (since it is redundant)

Nothing else needs to change in the arch-test (including the regenerating the tests), but ctg probably needs to be updated to reflect the actual name

allenjbaum, 2022-12-08:
We will fix this by deleting the TEST_RI version as part of splitting off the test macros

allenjbaum, 2023-02-09:
There are no further references to TEST_RI so it should be deleted

UmerShahidengr, 2024-10-31:
PR #545 has resolved this issue, will close this one after merging #545.

closed 2024-10-31 by UmerShahidengr