riscv/riscv-arch-test #279: Questions about RV32I verification

mole99, 2022-09-13:
Hi, I am currently trying to verify a minimal RISC-V core (RV32I).

I have tried to strip down my configurations as much as possible:

isa.yaml:

```
hart_ids: [0]
hart0:
  ISA: RV32I
  physical_addr_sz: 32
  User_Spec_Version: '2.3'
  supported_xlen: [32]
```

platform.yaml:

```
nmi:
  label: nmi_vector
reset:
  label: reset_vector
```

I run riscof with this command:

```
riscof run --config ./config.ini --suite ./riscv-arch-test/riscv-test-suite/rv32i_m --env ./riscv-arch-test/riscv-test-suite/env
```

Riscof then runs all the tests from `riscv-test-suite/rv32i_m/I/src` on both the DUT and the reference implementation, all of which pass. But then it also runs a single test called `riscv-test-suite/rv32i_m/privilege/src/misalign1-jalr-01.S` which fails.

My question is whether it is correct that the last test was selected, since it seems to test something from the privileged architecture?

Does my core need to implement Machine-Level ISA? I thought because Zicsr is optional Machine-Level ISA is optional too.

Is it also necessarry for a RISC-V compliant core to implement interrupts? Because in the `platform.yaml` I cannot remove `nmi` without errors.

Sorry for all the questions. I appreciate any help :)

pawks, 2022-09-13:
> Riscof then runs all the tests from `riscv-test-suite/rv32i_m/I/src` on both the DUT and the reference implementation, all of which pass. But then it also runs a single test called [...]

Yes, the test should be selected. It tests to verify the behavior of jalr instruction when the target address is set up to be at a `1` byte alignment(since the implementation should 0 out the lsb after calculating the target address before performing the redirection). 
It was placed in the privilege directory and the trap handler is enabled for the test only because it might be a common fault in the implementation where the lsb of the target in jalr is not set to 0. The conditions for this test should probably be updated to remove the trap handler and relocate the test to the `I` directory.

> Does my core need to implement Machine-Level ISA? I thought because Zicsr is optional Machine-Level ISA is optional too.

No, the core need not implement the Zicsr and the privilege ISA.

> Is it also necessarry for a RISC-V compliant core to implement interrupts? Because in the `platform.yaml` I cannot remove `nmi` without errors.

If the Zicsr and privilege haven't been implemented, then the interrupts are optional too. This looks like an oversight and should be fixed soon.

allenjbaum, 2022-09-13:
> Thank you for all the clarifications.
>
> From my side this issue can be closed. But I think it would be good to
> create two new issues to keep track. One for moving misalign1-jalr-01.S
> to I and another one for nmi in the yaml file.
>
> Should I go ahead and open them?

please do for tracking purposes. One would be in the arch-test repo, but
the nmi issue would be in riscv-config, I think?

mole99, 2022-09-14:
I have created #281 and riscv-software-src/riscv-config#103 to track the issues.
My questions have been answered therefore I will close this issue.

closed 2022-09-14 by mole99