riscv/riscv-arch-test #280: Missing information about the `RVTEST_CASE` argument format in test format spec.

pawks, 2022-09-13:
The arguments for the `RVTEST_CASE` are in a specific format and the test format spec needs to be updated to document the format. Something akin to [here](https://github.com/riscv-software-src/riscof/blob/a003752ff95dd5f9153c323dc2eb8ff7a0426dd9/docs/source/testformat.rst#rvtest_case-condition-formatting). Previously it was included in the riscof documentation along with the test format spec. But the test format spec was removed from riscof as it is actively maintained here.

allenjbaum, 2022-09-14:
I'll add it to the list of things that the new test_format spec will need
to document.

allenjbaum, 2022-12-08:
Close after updating test format spec (along with issue 190)

commit 99f106ba (2022-12-17): Fix #280. Added section in testformat spec for cond str of RVTEST_CASE macro.
PR #302 (2022-12-27): Canary patch
## Description
- Fix Canary definition to depend on sigalign.
- Fix SIGALIGN definition to work in all cases of FLEN and XLEN definitions.
- Fix inconsistencies in tests (missing trap regions and incomplete ISAs).
- Add Zicsr to ISA in RV32 privilege tests
- Modify signature size requirement to multiple of 4.

### Related Issues
- Fixes #190 
- Fixes #280
- Fixes #299
closed 2022-12-27 by neelgala